llvm/circt #4211: VERILATOR_ROOT ignored by check-circt-integration

tshadley, 2022-10-29:
found that if you don't have Verilator installed in $HOME/verilator, `ninja check-circt-integration` will fail looking for non-existent verilated.mk.

I tracked this to llvm-lit apparently needing a hardcoded list of environment variables to function properly:

https://github.com/llvm/llvm-project/blob/main/llvm/utils/lit/lit/TestingConfig.py#L24

So, adding VERILATOR_ROOT to this list is necessary. Am I missing something here? If there's a better place at which a fix should be applied, appreciate learning it.  Otherwise I'll close here and report this as an llvm issue.

mikeurbach, 2022-10-31:
CIRCT looks for verilator when cmake runs using the (nonstandard) VERILATOR_PATH environment variable: https://github.com/llvm/circt/blob/75c4a618491d0daca87d326afc074cdc3b48eef4/CMakeLists.txt#L211 If that's found and contains a verilator installation, it will populate this variable that our lit config uses: https://github.com/llvm/circt/blob/75c4a618491d0daca87d326afc074cdc3b48eef4/integration_test/lit.site.cfg.py.in#L41 So if you point VERILATOR_PATH to your installation, check-circt-integration should work fine. I expect if you look at the logs when you run cmake for CIRCT, you'll find "Did not find Verilator.", is that the case?

Another option is to use the bundled `utils/get-verilator.sh` script, which will install a known-good version of verilator to a path that the CIRCT cmake scripts check automatically, and re-run cmake.

tshadley, 2022-10-31:
@mikeurbach, many thanks for the hint.   I determined that VERILATOR_PATH was found correctly but it appears Verilator does not update its environment based on location of binary called but relies on configuration defaults during build (hence Vtop.mk has the wrong VERILATOR_ROOT).  I rebuilt Verilator with VERILATOR_ROOT set appropriately. 

To communicate a non-standard location for verilator to circt cmake through an environment variable, `$ENV{VERILATOR_PATH}` would be needed in find_program, but this is a tad off/confusing since the value would need to specify the `bin` directory, not the `verilator` executable (as is standard throughout for verilator_path).  

But using PATH as the environment variable instead -- `export PATH=$PATH:<nonstandard>/verilator/bin`works fine.

(Summary in short for using Verilator in a non-standard location: make sure verilator was built with environment variable `VERILATOR_ROOT` set correctly first and make sure that the non-standard location `bin` is in environment variable `PATH` before circt build.)

I'll rename issue to "Building circt with non-standard verilator location" and close.

closed 2022-10-31 by tshadley