llvm/circt #4252: Design with multiple modules using `--split-verilog` option outputs empty top file

carlosedp, 2022-11-04:
I have a design (sample from Chisel docs) with two modules. If I use the FirToolOptions below to split my modules into files, the Topmodule file ends-up empty.

```scala
//> using scala "2.13.8"
//> using lib "edu.berkeley.cs::chisel3::3.5.4"
//> using lib "com.sifive::chisel-circt::0.6.0"
//> using plugin "edu.berkeley.cs:::chisel3-plugin::3.5.4"

import chisel3._
import circt.stage.{ChiselStage, FirtoolOption}
import chisel3.stage.ChiselGeneratorAnnotation

class Foo extends Module {
  val io = IO(new Bundle {
    val a = Input(Bool())
    val b = Output(Bool())
  })
  io.b := !io.a
}

class FooWrapper extends RawModule {
  val a_i = IO(Input(Bool()))
  val b_o = IO(Output(Bool()))
  val clk = IO(Input(Clock()))
  val rstn = IO(Input(Bool()))

  val foo = withClockAndReset(clk, !rstn) { Module(new Foo) }

  foo.io.a := a_i
  b_o := foo.io.b
}

object Toplevel extends App {
  val chiselArgs =
    Array(
      "--target",
      "systemverilog",
      "--target-dir",
      "generated",
    )
  (new ChiselStage).execute(
    chiselArgs,
    Seq(
      ChiselGeneratorAnnotation(() => new FooWrapper),
      FirtoolOption("--split-verilog"),
      FirtoolOption("-o=generated"),
    ),
  )
}
```

Running:
```
> scala-cli FooWrapper.scala

Running CIRCT: 'firtool -format=fir -warn-on-unprocessed-annotations -verify-each=false --split-verilog -o=generated -disable-infer-rw -dedup -annotation-file generated/FooWrapper.anno.json < $input'
```

As a result, firtool generated in `generated` directory, 2 files, `Foo.sv` and `FooWrapper.sv` where the Toplevel module file, `FooWrapper` is blank. The `FooWrapper.fir` file has the contents showing Chisel output it correctly.

This doesn't happen when running without the FirToolOptions listed above (all in a single `sv` file where both modules end-up in `FooWrapper.sv`.

```
❯ firtool --version
LLVM (http://llvm.org/):
  LLVM version 16.0.0git
  Optimized build.
  Default target: x86_64-apple-darwin21.6.0
  Host CPU: skylake

CIRCT sifive/1/21/0
```

seldridge, 2022-11-05:
This was actually my bad with telling you to use `-split-verilog` and `-o=<output-directory>`. The problem is that `circt.stage.ChiselStage` can't handle that as it expects `firtool` to return a single file. This really needs a first class `ChiselStage` option. However, there is support for using an Annotation to do this.

If you instead use the following, you should be in business:

```scala
object Toplevel extends App {
  val chiselArgs =
    Array(
      "--target",
      "systemverilog",
      "--target-dir",
      "generated",
    )
  (new ChiselStage).execute(
    chiselArgs,
    Seq(
      ChiselGeneratorAnnotation(() => new FooWrapper),
      firrtl.EmitAllModulesAnnotation(classOf[firrtl.SystemVerilogEmitter])
    ),
  )
}
```

This `EmitAllModulesAnnotation` is a super terrible API, but it works. `circt.stage.ChiselStage` will see this, generate the appropriate `-split-verilog` option and then not overwrite the output file (which is what I think is what is happening).

carlosedp, 2022-11-05:
Awesome, It worked as expected (even tho it's kinda ugly 🫢)! 
Thanks Schuyler!

carlosedp, 2023-03-02:
Using latest Chisel 3.6-RC I managed to do:

```scala
    // Generate Verilog
    ChiselStage.emitSystemVerilogFile(
      new Toplevel(board, invreset, cpufreq),
      chiselArgs.value.toArray ++ Array("--split-verilog"),
      Array("--disable-all-randomization", "--strip-debug-info", "-lower-memories"),
    )
```

And split files were generated correctly. I think this can be closed...

closed 2023-03-02 by carlosedp