llvm/circt #4284: [ExportVerilog] Synchronous Negedge Reset?

leonardt, 2022-11-09:
Example MLIR
```
module attributes {circt.loweringOptions = "locationInfoStyle=none"} {
    hw.module @Foo(%I: i8, %CLK: i1, %RESETN: i1) -> (O: i8) {
        %1 = sv.reg {name = "Register_inst0"} : !hw.inout<i8>
        sv.alwaysff(posedge %CLK) {
            sv.passign %1, %I : i8
        } (syncreset : negedge %RESETN) {
            sv.passign %1, %2 : i8
        }
        %2 = hw.constant 0 : i8
        sv.initial {
            sv.bpassign %1, %2 : i8
        }
        %0 = sv.read_inout %1 : !hw.inout<i8>
        hw.output %0 : i8
    }
}
```

verilog out

```
module Foo(
  input  [7:0] I,
  input        CLK,
               RESETN,
  output [7:0] O);

  reg [7:0] Register_inst0;
  always_ff @(posedge CLK) begin
    if (RESETN)
      Register_inst0 <= 8'h0;
    else
      Register_inst0 <= I;
  end // always_ff @(posedge)
  initial
    Register_inst0 = 8'h0;
  assign O = Register_inst0;
endmodule
```

Investigating the docs and code it seems that negedge logic is only generated for async resets.  Is there any reason that synchronous negedge is not supported? If not, I'd be happy to update the code.

Relevant issue: https://github.com/phanrahan/magma/issues/1178

uenoku, 2022-11-10:
I think this is a bug of ExportVerilog but the comment says it intentionally restricts to async posedge reset. always_ff was introduced in https://github.com/llvm/circt/pull/491 and has not been changed much. 
@darthscsi do you remember why it is necessary to limit async reset?

leonardt, 2023-07-25:
Bumping this, if someone could provide a pointer to the relevant place in the code I'd be happy to contribute a change to support this.