llvm/circt #4300: [CI] GCC Debug LLVM build not caching

teqdruid, 2022-11-12:
GCC builds are taking forever because the llvm submodule build isn't being cached.

```
/usr/bin/tar --posix -z -cf cache.tgz -P -C /__w/circt/circt --files-from manifest.txt

gzip: stdout: No space left on device
/usr/bin/tar: cache.tgz: Wrote only 20[4](https://github.com/llvm/circt/actions/runs/3447357213/jobs/5753291283#step:26:4)[8](https://github.com/llvm/circt/actions/runs/3447357213/jobs/5753291283#step:26:8) of 10240 bytes
/usr/bin/tar: Child returned status 1
/usr/bin/tar: Error is not recoverable: exiting now
Warning: Tar failed with error: The process '/usr/bin/tar' failed with exit code 2
```

*sigh*

uenoku, 2022-11-15:
This [PR](https://github.com/llvm/circt/pull/4310) reduces the build size by changing gcc+debug to use shared lib (at least it passed the CI)

uenoku, 2022-11-16:
It looks cache works well. Closing..

closed 2022-11-16 by uenoku