llvm/circt #4318: CIRCT build failing at 'Building CXX object lib/Transforms/CMakeFiles/obj.CIRCTTransforms.dir/FlattenMemRefs.cpp.o'

rkshthrmsh, 2022-11-16:
While trying to build CIRCT according to the instructions here: https://circt.llvm.org/getting_started/, build fails on FlattenMemRefs.cpp.o. Please advice. LLVM build complete and tested.

`[1/21] Building CXX object lib/Transforms/CMakeFiles/obj.CIRCTTransforms.dir/FlattenMemRefs.cpp.o
FAILED: lib/Transforms/CMakeFiles/obj.CIRCTTransforms.dir/FlattenMemRefs.cpp.o 

...

ninja: build stopped: subcommand failed.`

The complete log is in the attached file

mikeurbach, 2022-11-16:
Looking at the error in the log, I'm wondering if you are using an older C/C++ compiler. What is your C/C++ compiler and version? Our CI tests build CIRCT with GCC 9.4 and Clang 13.0.1, so at least these are known to work. It might be that if you switch your compiler you can get around this issue.

rkshthrmsh, 2022-11-16:
Hi @mikeurbach, updating gcc to v9.4 helped finish the build, but check-circt failed. Should llvm be rebuilt with gcc 9.4? Please see the attached log.

[circt_build_log.txt](https://github.com/llvm/circt/files/10025630/circt_build_log.txt)

mikeurbach, 2022-11-16:
I think that attachment is the same contents as the original one?

Regardless, yes, you will probably have to recompile LLVM with the same gcc version. Sorry I didn't point this out earlier. 

I will go ahead and document the minimum supported compilers, and the need to use the same compiler for both LLVM and CIRCT, on the getting started guide.

rkshthrmsh, 2022-11-16:
Updated attachment. 

`gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~18.04) 9.4.0
`
`clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)`

Do you recommend updating clang to v13  before re-building llvm/mlir?

mikeurbach, 2022-11-16:
CMake should only find either GCC or LLVM, and which one you use is up to you. If you want to use GCC, using GCC 9.4 for both LLVM/MLIR and CIRCT compile should be fine. Or using Clang 13 for both. I think the important thing is to use the same compiler for both. We've seen weird issues with using one compiler for LLVM and a different one for CIRCT, and I'm not sure if those were resolved.

mikeurbach, 2022-11-16:
Great. I think some of those errors are related to the integration tests picking up your version of verilator, and that version not being compatible. I'm honestly not sure about the ESI failure. Depending on what you're trying to do with CIRCT, these might be safe to ignore.

rkshthrmsh, 2022-11-16:
What is the recommended version of verilator as per CI?

mikeurbach, 2022-11-16:
According to https://circt.llvm.org/getting_started/, we require at least v4.034. We also have a script to install Verilator for you: https://github.com/llvm/circt/blob/main/utils/get-verilator.sh. That installs v4.110, so it's possible we require at least v4.110 and the docs are out of date now. If you use that script, and re-run CMake for the CIRCT build, it should detect the installed verilator first instead of using your system verilator.

The newer version is required to use certain CIRCT features in ESI, but it might not be required depending on what you're trying to do. It's possible the older verilator consumes the CIRCT output for what you're trying to do.

rkshthrmsh, 2022-11-16:
Thanks, @mikeurbach! I am using verilator v5.002, so not quite sure what is causing the integration check to fail. Regardless, your suggestions solved the original issue.

closed 2022-11-16 by rkshthrmsh