llvm/circt #4343: [docs] FIRRTLAnnotation images broken on website

dtzSiFive, 2022-11-21:
https://circt.llvm.org/docs/Dialects/FIRRTL/FIRRTLAnnotations/#targets

Links to https://circt.llvm.org/docs/Dialects/FIRRTL/FIRRTLAnnotations/includes/img/firrtl-folded-module.png .
Image is currently placed here: https://circt.llvm.org/includes/img/firrtl-folded-module.png

Related: https://github.com/llvm/circt/pull/3338

teqdruid, 2023-01-23:
same think occurs on getting started page. the fix in #3338 isn't ideal.

matthew-humphrey, 2023-06-08:
There is a [stack overflow article](https://stackoverflow.com/questions/13808020/include-an-svg-hosted-on-github-in-markdown) on how to link to images within the markdown. The two techniques they mention are to use relative URLs or to utilize raw.github.com for the URL. It seems like for the Getting Started page, using the relative link would be a simple and effective way to fix this.

matthew-humphrey, 2023-06-08:
> @matthew-humphrey I think this issue is about what appears on circt.llvm.org, not what GitHub renders when viewing the markdown in the docs directory on GitHub. We use this markdown to generate the website via https://github.com/llvm/circt-www. Ideally, we'd have no broken links on the website (i.e. this issue should be resolved), and it would also be great for browsing the docs on GitHub to also work. I think using relative links works on GitHub, but caused issues on the website.

Thanks, @mikeurbach  - I missed that subtlety.