llvm/circt #4344: [PyCDE] Write a pycde install cmake target

teqdruid, 2022-11-21:
We currently use the "check-pycde" target when building the wheel. The new target should assemble all the necessary files and place them in the install directory to be packaged into the wheel.

mikeurbach, 2022-11-21:
I believe there should be a target named `install-PyCDE` already, can that be used?

teqdruid, 2022-11-21:
IIRC it's broken. Especially with the new BSP support which require aux
files.

mikeurbach, 2022-11-21:
I see. Well maybe it can be customized, since that seems like the "right" target to use for this.

teqdruid, 2022-11-21:
I agree that's the correct target.

teqdruid, 2024-07-31:
I think I took care of this awhile back such that building the PyCDE target works and `cmake --install --component PyCDE` works.

closed 2024-07-31 by teqdruid