llvm/circt #4369: [FIRRTL] IMCP produces incorrect results with aggregate

uenoku, 2022-11-29:
```scala
circuit Foo:
  module Bar:
    input in: UInt<1>[2]
    output out: UInt<1>
    wire a: UInt<1>
    a <= in[0]
    out <= a
  module Foo:
    input a: UInt<1>[2]
    output b: UInt<1>
    inst bar of Bar
    bar.in <= a
    b <= bar.out
```

`$ firtool -preserve-aggregate=1d-vec foo.fir`

produces the incorrect result:
```
module Bar(
  output out);

  assign out = 1'bx;
endmodule
```

darthscsi, 2022-11-30:
Can you dig into this further?  It shouldn't be possible to get an `x` on anything from firrtl.  IMCP certainly, no matter how buggy, shouldn't be causing an x assignment.

uenoku, 2022-11-30:
X is produced because currently IMCP deletes connection if the lattice state of its destination has unwritten state 
https://github.com/llvm/circt/blob/3363115a3e67c8bb89079f439985f5473e81f496/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp#L804

darthscsi, 2022-11-30:
An unwritten wire should produce a 'x'

darthscsi, 2022-11-30:
Which is my point.  What is making an unwritten wire get an 'x'?  Which isn't obviously what is happening, an output port is getting an 'x' (which shouldn't happen).

uenoku, 2022-11-30:
https://github.com/llvm/circt/blob/16b31d4eb6b1bb251cbbc93cbc2739b06bbfda4c/lib/Dialect/SV/SVOps.cpp#L1483-L1488

darthscsi, 2022-11-30:
That's a LowerToHW bug, arguably.  unwritten wires shouldn't exist in firrtl, but we do cause them in places.  The behavior is unspecified, but seems like it should be an indeterminate value situation, in which case it should be returning some constant.

darthscsi, 2022-11-30:
It's also a SV bug as unwritten wires are specified in the SV spec as having 'z' as a value.

darthscsi, 2022-11-30:
SV bug is fixed: 70cb3b0b520a17a8eec2c2c10a7f84a79224f9a0

PR #4370 (2022-11-30): [FIRRTL][IMCP] Mark aggregate as overdefined
https://github.com/llvm/circt/commit/fa2adf8109b3128cc8cb10e2b0f52a3f90502802 changed IMCP not to mark aggregate as overdefined in the initialization. That caused incorrect optimization so this PR marks aggregate as overdefined. 

Close #4369
commit 3363115a (2022-11-30): [FIRRTL][IMCP] Mark aggregate as overdefined (#4370)

https://github.com/llvm/circt/commit/fa2adf8109b3128cc8cb10e2b0f52a3f90502802 changed IMCP not to mark aggregate as overdefined in the initialization. That caused incorrect optimization so this PR marks aggregate as overdefined. 

Close #4369
closed 2023-03-08 by uenoku