llvm/circt #4378: [Calyx][ASAN] heap-use-after-free in lit tests

dtzSiFive, 2022-11-30:
Five lit tests, mostly SFCToCalyx, newly fail with heap-use-after-free.  cc #4353.

All error reports: https://gist.github.com/dtzSiFive/4561baec40ea6926de513c460cc6d19f .

First one is included below:
<details>
<summary>convert_pipeline.mlir</summary>

```
FAIL: CIRCT :: Conversion/PipelineToCalyx/convert_pipeline.mlir (490 of 494)
******************** TEST 'CIRCT :: Conversion/PipelineToCalyx/convert_pipeline.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /build/sifive/asan/bin/circt-opt /home/will/src/sifive/circt/test/Conversion/PipelineToCalyx/convert_pipeline.mlir -lower-static-logic-to-calyx -split-input-file | /build/sifive/asan/bin/FileCheck /home/will/src/sifive/circt/test/Conversion/PipelineToCalyx/convert_pipeline.mlir
--
Exit Code: 2

Command Output (stderr):
--
=================================================================
==1442208==ERROR: AddressSanitizer: heap-use-after-free on address 0x6080000019c4 at pc 0x000002379ef0 bp 0x7fff22ee1f90 sp 0x7fff22ee1f88
READ of size 4 at 0x6080000019c4 thread T0
    #0 0x2379eef in mlir::Operation::getResults() /home/will/src/sifive/circt/llvm/llvm/../mlir/include/mlir/IR/Operation.h:333:12
    #1 0x2379eef in mlir::Operation::use_empty() /home/will/src/sifive/circt/llvm/llvm/../mlir/include/mlir/IR/Operation.h:629:29
    #2 0x5beb22a in mlir::isOpTriviallyDead(mlir::Operation*) /home/will/src/sifive/circt/llvm/mlir/lib/Interfaces/SideEffectInterfaces.cpp:35:14
    #3 0x5865c88 in (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>) /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:211:11
    #4 0x5865c88 in mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:417:27
    #5 0x39389f7 in mlir::applyPatternsAndFoldGreedily(mlir::Operation*, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) /home/will/src/sifive/circt/llvm/llvm/../mlir/include/mlir/Transforms/GreedyPatternRewriteDriver.h:71:10
    #6 0x39389f7 in circt::pipelinetocalyx::PipelineToCalyxPass::runPartialPattern(mlir::RewritePatternSet&, bool) /home/will/src/sifive/circt/lib/Conversion/PipelineToCalyx/PipelineToCalyx.cpp:1488:11
    #7 0x39250b9 in circt::pipelinetocalyx::PipelineToCalyxPass::runOnOperation() /home/will/src/sifive/circt/lib/Conversion/PipelineToCalyx/PipelineToCalyx.cpp:1602:39
    #8 0x5b4f689 in mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:465:11
    #9 0x5b508d5 in mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:529:16
    #10 0x5b555a9 in mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:834:10
    #11 0x5b552ef in mlir::PassManager::run(mlir::Operation*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:814:60
    #12 0x46557b2 in performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, bool, bool) /home/will/src/sifive/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:91:17
    #13 0x4654f21 in processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, bool, bool, bool, bool, bool, bool, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, llvm::ThreadPool*) /home/will/src/sifive/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:139:12
    #14 0x4654f21 in mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool)::$_0::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) const /home/will/src/sifive/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:181:12
    #15 0x4654f21 in mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool)::$_0>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:12
    #16 0x46aaf96 in llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) const /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:12
    #17 0x46aace2 in mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0::operator()(llvm::StringRef) const /home/will/src/sifive/circt/llvm/mlir/lib/Support/ToolUtilities.cpp:88:16
    #18 0x46aa3b3 in void llvm::interleave<llvm::StringRef const*, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, void llvm::interleave<llvm::SmallVector<llvm::StringRef, 8u>, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, llvm::raw_ostream, llvm::StringRef>(llvm::SmallVector<llvm::StringRef, 8u> const&, llvm::raw_ostream&, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, llvm::StringRef const&)::'lambda'(), void>(llvm::SmallVector<llvm::StringRef, 8u>, llvm::SmallVector<llvm::StringRef, 8u>, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, llvm::raw_ostream) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:2045:3
    #19 0x46aa3b3 in void llvm::interleave<llvm::SmallVector<llvm::StringRef, 8u>, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, llvm::raw_ostream, llvm::StringRef>(llvm::SmallVector<llvm::StringRef, 8u> const&, llvm::raw_ostream&, mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool)::$_0, llvm::StringRef const&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:2067:3
    #20 0x46aa3b3 in mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ [... cut ...]
    #35 0x465365e in mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) /home/will/src/sifive/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:306:14
    #36 0x1f24651 in main /home/will/src/sifive/circt/tools/circt-opt/circt-opt.cpp:70:23
    #37 0x7feae7e2924d in __libc_start_call_main (/nix/store/ynn1by1qdl16q6qwwh2h7zkgrn36c6i8-glibc-2.35-163/lib/libc.so.6+0x2924d) (BuildId: 1433936b3a4bd2e8bc1129b0563f1c826138656f)
    #38 0x7fff22ee72ee  ([stack]+0x202ee)

```
</details>

mikeurbach, 2022-11-30:
Thanks for filing this. This might be related to the terribly hacky thing we did to make this pass work after the LLVM bump, but I have a hunch it's due to the change to the greedy pattern driver in the LLVM bump.

dtzSiFive, 2022-12-05:
Thanks for the insight, that seems likely.  No blame intended just linking related issue for context and since folks working on that likely have insights here.  cc #4381, the resolution of which is very likely to address this as well.

teqdruid, 2023-01-25:
This (we strongly suspect) is causing tests to fail and blocking an llvm submodule bump. As discussed in the ODM today, we are XFAIL'ing the following tests:

```
Failed Tests (6):
  CIRCT :: Conversion/PipelineToCalyx/convert_pipeline.mlir
  CIRCT :: Conversion/SCFToCalyx/cider_source_location.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_simple.mlir
  CIRCT :: Conversion/SCFToCalyx/errors.mlir
```

closed 2023-09-09 by dtzSiFive