llvm/circt #4382: [Windows] `Bindings/Python/dialects/sv.py` integration test crashes

teqdruid, 2022-12-01:
```
> C:/Python/Python39/python.exe C:\code\circt\integration_test\Bindings\Python\dialects\sv.py
Assertion failed: !PyErr_Occurred(), file C:\Python\Python39\Lib\site-packages\pybind11\include\pybind11\detail/common.h, line 940
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Exception Code: 0x80000003
 #0 0x00007ffd1d5ac215 PyInit__circt (C:\code\circt_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0x3c215)
 #1 0x00007ffd7527e5f5 (C:\WINDOWS\System32\ucrtbase.dll+0x7e5f5)
 #2 0x00007ffd7527f601 (C:\WINDOWS\System32\ucrtbase.dll+0x7f601)
 #3 0x00007ffd7528106a (C:\WINDOWS\System32\ucrtbase.dll+0x8106a)
 #4 0x00007ffd75280f5d (C:\WINDOWS\System32\ucrtbase.dll+0x80f5d)
 #5 0x00007ffd752811f1 (C:\WINDOWS\System32\ucrtbase.dll+0x811f1)
 #6 0x00007ffd1d5725ee pybind11::reference_cast_error::reference_cast_error(char const *) (C:\code\circt_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0x25ee)
 #7 0x00007ffd1d5a7c92 PyInit__circt (C:\code\circt_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0x37c92)
 #8 0x00007ffd1d5a7cb4 PyInit__circt (C:\code\circt_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0x37cb4)
 #9 0x00007ffd1d57b6ae pybind11::error_already_set::operator=(class pybind11::error_already_set &&) (C:\code\circt_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0xb6ae)
#10 0x00007ffcef398a8a cfunction_call D:\_w\1\s\Objects\methodobject.c:554:0
#11 0x00007ffcef38069a _PyObject_Call D:\_w\1\s\Objects\call.c:283:0
#12 0x00007ffcef345215 _PyObject_MakeTpCall D:\_w\1\s\Objects\call.c:192:0
#13 0x00007ffcef49ba1c _guard_xfg_dispatch_icall_nop D:\_w\1\s\Objects\descrobject.c:1575:0
#14 0x00007ffcef34e76a _PyObject_GenericGetAttrWithDict D:\_w\1\s\Objects\object.c:1201:0
#15 0x00007ffcef34f6cd _PyEval_EvalFrameDefault D:\_w\1\s\Python\ceval.c:2996:0
#16 0x00007ffcef34d06f _PyEval_EvalCode D:\_w\1\s\Python\ceval.c:4329:0
#17 0x00007ffcef3db0ed _PyEval_EvalCodeWithName D:\_w\1\s\Python\ceval.c:4368:0
#18 0x00007ffcef3db02f PyEval_EvalCodeEx D:\_w\1\s\Python\ceval.c:4384:0
#19 0x00007ffcef3daf8d PyEval_EvalCode D:\_w\1\s\Python\ceval.c:834:0
#20 0x00007ffcef3dab8e run_eval_code_obj D:\_w\1\s\Python\pythonrun.c:1222:0
#21 0x00007ffcef3dab0e run_mod D:\_w\1\s\Python\pythonrun.c:1243:0
#22 0x00007ffcef35b94f pyrun_file D:\_w\1\s\Python\pythonrun.c:1140:0
#23 0x00007ffcef35b6d4 pyrun_simple_file D:\_w\1\s\Python\pythonrun.c:453:0
#24 0x00007ffcef35c52b PyRun_SimpleFileExFlags D:\_w\1\s\Python\pythonrun.c:484:0
#25 0x00007ffcef35b084 PyRun_AnyFileExFlags D:\_w\1\s\Python\pythonrun.c:93:0
#26 0x00007ffcef35aa0a pymain_run_file D:\_w\1\s\Modules\main.c:373:0
#27 0x00007ffcef3177a4 pymain_run_python D:\_w\1\s\Modules\main.c:599:0
#28 0x00007ffcef31762d Py_RunMain D:\_w\1\s\Modules\main.c:679:0
#29 0x00007ffcef45d5e1 Py_Main D:\_w\1\s\Modules\main.c:720:0
#30 0x00007ff734621254 __scrt_common_main_seh d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288:0
#31 0x00007ffd7697244d (C:\WINDOWS\System32\KERNEL32.DLL+0x1244d)
#32 0x00007ffd77e2dfb8 (C:\WINDOWS\SYSTEM32\ntdll.dll+0x5dfb8)
```

I've tracked this crash to [this line in the test](https://github.com/llvm/circt/blob/04898c0fc44ccab94595fcf315643563c6fdfdbe/integration_test/Bindings/Python/dialects/sv.py#L14). If I comment it out, the test doesn't crash.

teqdruid, 2022-12-01:
I'm guessing this wasn't ASAN clean. It'd be useful to have ASAN checks on the integration tests as well. (Especially since valgrind is worthless on python.)

closed 2022-12-01 by teqdruid

mikeurbach, 2022-12-01:
How old is this bit of code? I thought everything was ASAN clean about two weeks ago... anyway, agreed we should have ASAN in all CI.

teqdruid, 2022-12-01:
It's pretty old. Certainly older than a few weeks.

commit 44a84371 (2022-12-01): [Python] SVAttributeAttr.get(): need temporary storage for expr

Fixes #4382. The `std::string` casted to from the expression gets
destructed before it is used. Store it somewhere to avoid that
destruction until the function exits.

I'm guessing this wasn't ASAN clean.