llvm/circt #4387: [PrepareForEmission] Constant array async reset values create a wire

uenoku, 2022-12-01:
Since HW doesn't have constant aggregate operations, PrepareForEmission sometimes fails to recognize the value to be constant and creates a temporary wire for it.
```
circuit Foo:
  module Foo:
    input b: UInt<1>[2]
    input clock: Clock
    input reset: AsyncReset
    output a: UInt<1>[2]
    wire namedAsyncResetValue: UInt<1>[2]
    namedAsyncResetValue[0] <= UInt<1>(0)
    namedAsyncResetValue[1] <= UInt<1>(0)
    reg r : UInt<1>[2], clock with :
      reset => (reset, namedAsyncResetValue)
    r <= b
    a <= r
```

```verilog
 wire [1:0] namedAsyncResetValue = {2{1'h0}};
  reg  [1:0] r;
  always @(posedge clock or posedge reset) begin
    if (reset)
      r <= namedAsyncResetValue;
    else
      r <= b;
```

uenoku, 2022-12-01:
A clean solution would be to introduce HW aggregate constant.

youngar, 2022-12-09:
With https://github.com/llvm/circt/pull/4402, compiling this example using `firtool --preserve-aggregate=1d-vec --preserve-public-types=0` gives the following:
```verilog
  reg [1:0] r;
  always @(posedge clock or posedge reset) begin
    if (reset)
      r <= '{1'h0, 1'h0};
    else
      r <= b;
  end // always @(posedge, posedge)
```

uenoku, 2022-12-09:
Closed by #4402. Thanks!

closed 2022-12-09 by uenoku