llvm/circt #4391: [InnerSymDCE] inner symbols on ports are not DCE'd

dtzSiFive, 2022-12-01:
Module ports are not considered for removal in InnerSymDCE, such as this example:
```
firrtl.circuit "Simple" {

  firrtl.module @Simple() {
    %out = firrtl.instance child sym @child @Child(out out : !firrtl.uint<1>)
  }
  firrtl.module @Child(out %out : !firrtl.uint<1> sym @outsym) {
    %c0_ui1 = firrtl.constant 0: !firrtl.uint<1>
    firrtl.strictconnect %out, %c0_ui1 : !firrtl.uint<1>
  }
}
```

Where `@outsym` is dead and can be removed.

mikeurbach, 2022-12-01:
Good catch. InnerSymDCE is currently only looking at ops within modules that have an `inner_sym`: https://github.com/llvm/circt/blob/5240cd4610246796c045fc5456d76b5482cefea1/lib/Dialect/FIRRTL/Transforms/InnerSymbolDCE.cpp#L71-L76

It could also check the `moduleOp` port attributes.

dtzSiFive, 2022-12-01:
Inner symbol fields are not handled either, as mentioned in that comment on the PR, here's a small extension to the above example to demonstrate:

```mlir
firrtl.circuit "Simple" {
  firrtl.module @Simple() {
    %out = firrtl.instance child sym @child @Child(out out : !firrtl.uint<1>)
  }
  firrtl.module @Child(out %out : !firrtl.uint<1> sym @outsym) {
    %c0_ui1 = firrtl.constant 0: !firrtl.uint<1>
    firrtl.strictconnect %out, %c0_ui1 : !firrtl.uint<1>
    %w3 = firrtl.wire sym [<@x,1,public>] : !firrtl.vector<uint<1>,1>
  }
  firrtl.hierpath private @nla [@Simple::@child, @Child::@x]
}
```

The NLA is broken by InnerSymDCE presently:

```
inner-symbol-dce-field.mlir:14:3: error: 'firrtl.hierpath' op  operation with symbol: #hw.innerNameRef<@Child::@x> was not found 
  firrtl.hierpath private @nla [@Simple::@child, @Child::@x]
  ^
inner-symbol-dce-field.mlir:14:3: note: see current operation: "firrtl.hierpath"() {namepath = [#hw.innerNameRef<@Simple::@child>, #hw.innerNameRef<@Child::@x>], sym_name = "nla", sym_visibility = "private"} : () -> ()
```

PR #4787 (2023-03-24): Fix InnerSymDCE, add support for erasing+setting inner symbols
Fixes #4391 .

FIRRTL-only version of #4418.
closed 2023-03-24 by dtzSiFive