llvm/circt #4399: [FIRRTL][ExportingVerilog]Generated Verilog file is abnormally large.

Siudya, 2022-12-03:
CIRCT source code tag: sifive/1/23/0
OS: Centos 7.9
Generated Verilog file is extremely large in some unkown cases. Here is how to repete:

```
git clone https://github.com/OpenXiangShan/XiangShan.git
cd XiangShan
git checkout nanhu-v2
make init
mill -i XiangShan.runMain top.TopMain -td build --config DefaultConfig --full-stacktrace -X none -E chirrtl -o XSTop.chirrtl.fir --num-cores 2 --fpga-platform --disable-all --remove-assert
firtool --disable-all-randomization --verilog --disable-annotation-unknown --dedup --format=fir --annotation-file=build/XSTop.anno.json -o build/XSTop.v build/XSTop.chirrtl.fir
```
The generated XSTop.v is 3.9GiB，which is 150MiB when using default firrtl compiler.

The module Ftq may be where the problem is. The size of it's verilog file is 3.8GiB, which is around 9MiB when using the default scala firrtl compiler.

seldridge, 2022-12-03:
Interesting. Thanks for the report.

Would you be able to provide the input `build/XStop.chirrtl.fir` and `build/XSTop.anno.json` files? I'm having a little trouble building this:

1. I think I need to create a `build/` directory to get around:

```
Generating MBIST_Core.csv
error
Exception in thread "main" java.io.FileNotFoundException: build/MBIST_Core.csv (No such file or directory)
```

2. Running again ran into some issue after this point:

```
Generating MBIST_L3S0.csv
Generating MBIST_L3S1.csv
Generating MBIST_L3S2.csv
Generating MBIST_L3S3.csv
1 targets failed
XiangShan.runMain subprocess failed
```

Siudya, 2022-12-04:
> Interesting. Thanks for the report.
> 
> Would you be able to provide the input `build/XStop.chirrtl.fir` and `build/XSTop.anno.json` files? I'm having a little trouble building this:
> 
>     1. I think I need to create a `build/` directory to get around:
> 
> 
> ```
> Generating MBIST_Core.csv
> error
> Exception in thread "main" java.io.FileNotFoundException: build/MBIST_Core.csv (No such file or directory)
> ```
> 
>     2. Running again after creating a build directory I ran into some issue after this point:
> 
> 
> ```
> Generating MBIST_L3S0.csv
> Generating MBIST_L3S1.csv
> Generating MBIST_L3S2.csv
> Generating MBIST_L3S3.csv
> 1 targets failed
> XiangShan.runMain subprocess failed
> ```

[bug_report.tar.txt](https://github.com/llvm/circt/files/10148973/bug_report.tar.txt)
Generating the .fir file needs more than 32GiB RAM while 64GiB is better.
Here is the generated files. Please rename the suffix of the attachment to ".tar.bz2".

seldridge, 2022-12-04:
I can reproduce. Yes, the coding style in `NewFTQ.scala` is causing problems. This is creating one incredibly long if/else chain which results it in being, by-far, the largest output Verilog module:

```
# l -S issues/4399/mfc       
total 3200480
-rw-r--r--     1 schuylere  staff   1.4G Dec  4 17:14 Ftq.sv
-rw-r--r--     1 schuylere  staff   8.3M Dec  4 16:57 XSCore.sv
-rw-r--r--     1 schuylere  staff   4.4M Dec  4 16:57 LoadQueue.sv
```

All the assignments to `commitStateQueue` seem to be causing some issue where CIRCT is preserving these as opposed to canonicalization them into a simpler structure. This `commitStateQueue` looks like a 2-bit vector of size 64x16. This is then assigned in a several different blocks using subaccesses. Just one of these vector elements produces ~16k different assignments in the output verilog. This can likely be fixed with a different coding style inside `NewFTQ.scala`. However, this also indicates that CIRCT should do better on this code example.

An example of the Verilog output for one vector entry is the following. Somebody might have an idea on exactly what is missing here:

```verilog
          if (_GEN_19243 & _GEN_20345 | _GEN_19245)     // NewFtq.scala:1005:63, :1009:71
            commitStateQueue_0_0 <= 2'h2;       // Mux.scala:47:70, NewFtq.scala:565:33
          else if (io_fromBackend_rob_commits_4_valid) begin
            if (_T_459) begin   // NewFtq.scala:1008:31
              if (_GEN_17029)   // NewFtq.scala:1005:63, :1009:71
                commitStateQueue_0_0 <= 2'h2;   // Mux.scala:47:70, NewFtq.scala:565:33
              else if (io_fromBackend_rob_commits_3_valid) begin
                if (_T_451) begin       // NewFtq.scala:1008:31
                  if (_GEN_13721)       // NewFtq.scala:1005:63, :1009:71
                    commitStateQueue_0_0 <= 2'h2;       // Mux.scala:47:70, NewFtq.scala:565:33
                  else if (io_fromBackend_rob_commits_2_valid) begin
                    if (_T_443) begin   // NewFtq.scala:1008:31
                      if (_GEN_10413)   // NewFtq.scala:1005:63, :1009:71
                        commitStateQueue_0_0 <= 2'h2;   // Mux.scala:47:70, NewFtq.scala:565:33
                      else if (io_fromBackend_rob_commits_1_valid) begin
                        if (_T_435) begin       // NewFtq.scala:1008:31
                          if (_GEN_7105)        // NewFtq.scala:1005:63, :1009:71
                            commitStateQueue_0_0 <= 2'h2;       // Mux.scala:47:70, NewFtq.scala:565:33
                          else if (io_fromBackend_rob_commits_0_valid) begin
                            if (_T_427) begin   // NewFtq.scala:1008:31
                              if (_GEN_3797)    // NewFtq.scala:1005:63, :1009:71
                                commitStateQueue_0_0 <= 2'h2;   // Mux.scala:47:70, NewFtq.scala:565:33
                              else if (_GEN_1670)       // NewFtq.scala:978:51, :989:19, :991:60, :992:13
                                commitStateQueue_0_0 <= 2'h0;   // NewFtq.scala:565:33
                              else if (_GEN_579)        // NewFtq.scala:806:23, :811:{32,37}
                                commitStateQueue_0_0 <= 2'h1;   // Mux.scala:47:70, NewFtq.scala:565:33
                              else if (_GEN_389)        // 610:54:611:49, NewFtq.scala:565:33, NewFtq.scala 606:17:565:33
                                commitStateQueue_0_0 <= 2'h0;   // NewFtq.scala:565:33
                            end
                            else if (_T_430) begin      // NewFtq.scala:1010:36
                              if (_GEN_4836)    // NewFtq.scala:1005:63, :1011:71
                                commitStateQueue_0_0 <= 2'h2;   // Mux.scala:47:70, NewFtq.scala:565:33
                              else if (_GEN_1670)       // NewFtq.scala:978:51, :989:19, :991:60, :992:13
                                commitStateQueue_0_0 <= 2'h0;   // NewFtq.scala:565:33
                              else if (_GEN_579)        // NewFtq.scala:806:23, :811:{32,37}
                                commitStateQueue_0_0 <= 2'h1;   // Mux.scala:47:70, NewFtq.scala:565:33
                              else if (_GEN_389)        // 610:54:611:49, NewFtq.scala:565:33, NewFtq.scala 606:17:565:33
                                commitStateQueue_0_0 <= 2'h0;   // NewFtq.scala:565:33
                            end
                            else if (_GEN_5042) // NewFtq.scala:1005:63, :1012:45, :1014:36
                              commitStateQueue_0_0 <= 2'h2;     // Mux.scala:47:70, NewFtq.scala:565:33
                            else if (_GEN_1670) // NewFtq.scala:978:51, :989:19, :991:60, :992:13
                              commitStateQueue_0_0 <= 2'h0;     // NewFtq.scala:565:33
                            else if (_GEN_579)  // NewFtq.scala:806:23, :811:{32,37}
                              commitStateQueue_0_0 <= 2'h1;     // Mux.scala:47:70, NewFtq.scala:565:33
                            else if (_GEN_389)  // 610:54:611:49, NewFtq.scala:565:33, NewFtq.scala 606:17:565:33
                              commitStateQueue_0_0 <= 2'h0;     // NewFtq.scala:565:33
                          end
                          else if (_GEN_1670)   // NewFtq.scala:978:51, :989:19, :991:60, :992:13
                            commitStateQueue_0_0 <= 2'h0;       // NewFtq.scala:565:33
                          else if (_GEN_579)    // NewFtq.scala:806:23, :811:{32,37}
[... cut ...]
                        else if (_GEN_1670)     // NewFtq.scala:978:51, :989:19, :991:60, :992:13
                          commitStateQueue_0_0 <= 2'h0; // NewFtq.scala:565:33
                        else if (_GEN_579)      // NewFtq.scala:806:23, :811:{32,37}
                          commitStateQueue_0_0 <= 2'h1; // Mux.scala:47:70, NewFtq.scala:565:33
                        else if (_GEN_389)      // 610:54:611:49, NewFtq.scala:565:33, NewFtq.scala 606:17:565:33
                          commitStateQueue_0_0 <= 2'h0; // NewFtq.scala:565:33
                      end
```

seldridge, 2022-12-04:
Alright, I think I have a somewhat minimal example. The problem is in the `SeqFIRRTLToSV` pass. This causes a massive blowup in IR size. A FIRRTL-level example is the following:

```
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    input a : UInt<1>[16]
    input index : UInt<2>
    output b : UInt<1>[16]

    reg r : UInt<2>[16], clock with :
      reset => (UInt<1>("h0"), r) 
    when a[0] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[1] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[2] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[3] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[4] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[5] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[6] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[7] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[8] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[9] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[10] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[11] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[12] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[13] : 
      r[index] <= UInt<1>("h1") 
    else :
      r[index] <= UInt<2>("h2") 
    when a[14] : 
[... 207 lines cut ...]
    %181 = comb.mux bin %a_14, %174, %180 : i2
    %182 = comb.mux bin %6, %c-2_i2, %171 : i2
    %183 = comb.mux bin %a_14, %175, %182 : i2
    %184 = comb.mux bin %0, %c1_i2, %177 : i2
    %185 = comb.mux bin %2, %c1_i2, %179 : i2
    %186 = comb.mux bin %4, %c1_i2, %181 : i2
    %187 = comb.mux bin %6, %c1_i2, %183 : i2
    %188 = comb.mux bin %0, %c-2_i2, %177 : i2
    %189 = comb.mux bin %a_15, %184, %188 : i2
    %190 = comb.mux bin %2, %c-2_i2, %179 : i2
    %191 = comb.mux bin %a_15, %185, %190 : i2
    %192 = comb.mux bin %4, %c-2_i2, %181 : i2
    %193 = comb.mux bin %a_15, %186, %192 : i2
    %194 = comb.mux bin %6, %c-2_i2, %183 : i2
    %195 = comb.mux bin %a_15, %187, %194 : i2
    %196 = comb.extract %r_0 from 0 : (i2) -> i1
    %197 = comb.extract %r_1 from 0 : (i2) -> i1
    %198 = comb.extract %r_2 from 0 : (i2) -> i1
    %199 = comb.extract %r_3 from 0 : (i2) -> i1
    hw.output %196, %197, %198, %199, %false, %false, %false, %false, %false, %false, %false, %false, %false, %false, %false, %false : i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1, i1
  }
}
```

After this one pass, this is 200MiB of `.mlir`.

Siudya, 2022-12-05:
Indeed, I can reproduce with your minimal case. These structures can be summarized as follows: 
A big matrix or vector consists of small elements, each of which is with a mux chain (an independent write condition may be just enough ?) connected to its D port.
Such a structure is neccessary in our design when storing meta data written by multiple paths. 
According to your example, it seems that even a small vector with respectively few and short write paths to its elements can trigger this problem. It brings difficulties to physical design and RTL verification.
Is there any workaround I can try, or I have to keep using SFC for now?

seldridge, 2022-12-05:
I don't have a workaround right now. 😅 This is clearly a bug... I expect this will get fixed this week.

We've seen problems with this pass before and had to back out changes to it. Relevant commits:
- 101115452e7ceba6c00d5fc5f02541c24abb2719
- 6380c594bfa35cb3aec35800e0bb7fe6088c1bf4

I think @nandor or @darthscsi may have an idea on a course of action.

uenoku, 2022-12-05:
The exponential IR is caused by constructing if-else trees for mux register updates. SFC does the same thing but it seems SFC restricts the mux depth:
https://github.com/chipsalliance/firrtl/blob/4f3d1003811aa38d10e32b347c8607414d9be034/src/main/scala/firrtl/transforms/FlattenRegUpdate.scala#L86

darthscsi, 2022-12-05:
Small canonicalizer and it reduces to:
```
  always @(posedge clock) begin
    automatic logic [1:0] _GEN = a_15 ? 2'h1 : 2'h2;
    r_0 <= index == 2'h0 ? _GEN : r_0;
    r_1 <= index == 2'h1 ? _GEN : r_1;
    r_2 <= index == 2'h2 ? _GEN : r_2;
    r_3 <= (&index) ? _GEN : r_3;
  end // always @(posedge)
```

darthscsi, 2022-12-05:
with https://github.com/llvm/circt/pull/4405 the test case is 89M
```
~/sifive/circt/tmp/bug_report$ du -hs v2
96M	v2
```

darthscsi, 2022-12-05:
Seems pretty well addressed by #4403 and #4405

closed 2022-12-05 by darthscsi