llvm/circt #4400: [ExportVerilog] How can I disable hierarchy flattening?

Siudya, 2022-12-05:
CIRCT source code tag: sifive/1/22/1
OS: CentOS 7.9
Docker: 18.06.3-ce
CIRCT image is built from Ubuntu 20.04
```
firtool --verilog --disable-annotation-unknown --dedup --format=fir --annotation-file=XSTop.anno.json -o XSTop.v XSTop.chirrtl.fir 
```
The module ctrlBlock and its submodules is flattened, which we do not expect. We hope flattening and hardening module in physical design stage with certain EDA tools, so this is an over-optimization for us.
Is it possible to disable module flattening or keep modules hierarchy when we convert FIR codes to RTL codes with CIRCT?[sources.tar.txt](https://github.com/llvm/circt/files/10150373/sources.tar.txt)(Please rename the suffix of the attachment to ".tar.bz2")

seldridge, 2022-12-05:
I think this is user error. That module has an `InlineAnnotation` on it. All `InlineAnnotation` and `FlattenAnnotation`s are automatically resolved by `firtool`. (The module inliner pass _always_ runs, but it can be disabled with `-disable-inliner`):

```
# cat bug_report/XSTop.anno.json | jq -c '.[] | select ( .class | contains("Inline")) | .target' | sort 
"XSTop.AXI4Deinterleaver_1"
"XSTop.CtrlBlock"
"XSTop.DifftestArchEvent"
"XSTop.DifftestArchFpRegState"
"XSTop.DifftestArchIntRegState"
"XSTop.DifftestBasicInstrCommit"
"XSTop.DifftestBasicTrapEvent"
"XSTop.DifftestCSRState"
"XSTop.DifftestDebugMode"
"XSTop.DifftestFpWriteback"
"XSTop.DifftestIntWriteback"
"XSTop.DifftestLrScEvent"
"XSTop.Dispatch2Rs"
"XSTop.Dispatch2Rs_1"
"XSTop.Dispatch2Rs_2"
"XSTop.ExuBlock"
"XSTop.ExuBlock_1"
"XSTop.ReservationStationWrapper"
"XSTop.ReservationStationWrapper_1"
"XSTop.ReservationStationWrapper_2"
"XSTop.ReservationStationWrapper_3"
"XSTop.ReservationStationWrapper_4"
"XSTop.ReservationStationWrapper_5"
"XSTop.ReservationStationWrapper_6"
"XSTop.ReservationStationWrapper_7"
"XSTop.Rob"
"XSTop.STD_CLKGT_func"
"XSTop.Scheduler"
"XSTop.Scheduler_1"
"XSTop.TLWidthWidget_1"
"XSTop.TLWidthWidget_2"
"XSTop.Wb2Ctrl"
"XSTop.WbArbiter"
"XSTop.WbArbiterWrapper"
"XSTop.WbArbiter_1"
```

To resolve this, whatever is creating this annotation should be removed.

seldridge, 2022-12-05:
To further clarify: there is no cost-based, automatic module/instance inlining. Any inlining is user-directed via `InlineAnnotation` or `FlattenAnnotation`.