llvm/circt #4411: [CI][Windows] Build failure, missing ZLIB::ZLIB

dtzSiFive, 2022-12-06:
Latest commit failed Windows CI: https://github.com/llvm/circt/actions/runs/3630835375/jobs/6124670021

Here's the error:
```
 CMake Error at llvm/install/lib/cmake/llvm/LLVMExports.cmake:64 (set_target_properties):
  The link interface of target "LLVMSupport" contains:

    ZLIB::ZLIB
-- Generating done

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  llvm/install/lib/cmake/llvm/LLVMConfig.cmake:233 (include)
  llvm/install/lib/cmake/mlir/MLIRConfig.cmake:10 (find_package)
  CMakeLists.txt:65 (find_package)

CMake Generate step failed.  Build files cannot be regenerated correctly.
ninja: error: loading 'build.ninja': The system cannot find the file specified.

ninja: error: loading 'build.ninja': The system cannot find the file specified.
```

Commit it failed on seems unrelated.

teqdruid, 2022-12-06:
Running another build got through that step...

teqdruid, 2022-12-06:
it found the zlib library: `-- Found ZLIB: C:/Strawberry/c/lib/libz.a (found version "1.2.11")` so it wasn't a build image change as I feared. Since this isn't (easily) reproducible, I'm going to consider it a transitory failure. Let's keep this issue open for the next week or so in case it occurs again.

dtzSiFive, 2023-02-17:
This seems to be working now, and closing as mentioned.  :crossed_fingers: