llvm/circt #4446: [FIRRTL] Non-deterministic test failures on Windows

teqdruid, 2022-12-14:
I've recently setup an internal CI flow which builds CIRCT on a pretty beefy VM -- Windows and Linux. At some point in the last week (or so), some FIRRTL tests started failing with this signature:

```
FAIL: CIRCT :: firtool/split-verilog.mlir (449 of 496)
******************** TEST 'CIRCT :: firtool/split-verilog.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   c:\__w\1\b\build\bin\firtool.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --format=mlir -verilog | c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG
: 'RUN: at line 2';   rm -rf C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp
: 'RUN: at line 3';   c:\__w\1\b\build\bin\firtool.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --format=mlir -split-verilog -o=C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp
: 'RUN: at line 4';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-FOO < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/foo.sv
: 'RUN: at line 5';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-BAR < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/bar.sv
: 'RUN: at line 6';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-USB < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/usb.sv
: 'RUN: at line 7';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-INOUT-3 < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/inout_3.sv
: 'RUN: at line 8';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-CUSTOM-1 < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/custom1.sv
: 'RUN: at line 9';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=VERILOG-CUSTOM-2 < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/custom2.sv
: 'RUN: at line 10';   c:\__w\1\b\build\bin\filecheck.exe C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir --check-prefix=LIST < C:\__w\1\b\build\tools\circt\test\firtool\Output\split-verilog.mlir.tmp/filelist.f
--
Exit Code: 3221225477

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "c:\__w\1\b\build\bin\firtool.exe" "C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir" "--format=mlir" "-verilog"
note: command had no output on stdout or stderr
error: command failed with exit status: 0xc0000005
$ "c:\__w\1\b\build\bin\filecheck.exe" "C:\__w\1\s\CIRCT\circt\test\firtool\split-verilog.mlir" "--check-prefix=VERILOG"
```

It's different tests each time. I've run three builds (2x Release, 1x RelWithDebugInfo) and seen the following tests fail:
```
  CIRCT :: firtool/connect.fir
  CIRCT :: firtool/extract-test-code.fir
  CIRCT :: firtool/firtool.fir
  CIRCT :: firtool/prefixMemory.fir
  CIRCT :: firtool/split-verilog.mlir
  CIRCT :: Dialect/FIRRTL/SFCTests/complex-asserts.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/emit-metadata.fir
```

Unfortunately, debug builds run out of disk space. (Thank you so much MSVC for being such a disk hog.)

There are two differences which come to mind relative to the CIRCT repo Windows CI builds:
- This one is a unified build.
- This one runs on a 16 core machine

Since different tests are failing for the same commit, this smells like a multithreading problem. I seem to recall this happening in the past on Windows: https://github.com/llvm/circt/issues/376#issuecomment-887919073

teqdruid, 2022-12-14:
I'm leaving on vacation after tomorrow, so I probably won't have time to debug this further until after Christmas, maybe next year.

teqdruid, 2022-12-14:
Actually, it was pretty easy to get some more space and compile in debug mode. In debug mode, I'm seeing other tests fail due to being out of memory. ([log](https://github.com/llvm/circt/files/10225500/48.txt). I may just be running out of memory and need to throttle back test parallelism.

Hold please.

teqdruid, 2022-12-15:
Reducing the lit parallelism fixed the problem... which implies to me that our tests use more memory than the LLVM compilation? A `-j16` unified build runs fine, but a `-j16` lit run of the circt tests causes memory allocation errors? That just seems wrong. I'm running this in a Windows docker container (which may or may not have a relatively low memory limit on it), but I'm running the build in the same container, sooo...

Is there an easy way to get peak memory usage for each of our tests?

seldridge, 2023-11-11:
Linking this to the most recent variant of this: https://github.com/llvm/circt/issues/6193

It seems to help to run on the newest Windows runner (2022). I'll close this as this is probably too old to debug. However, the comments and link to the latest issue should allow our future selves to find your parallelism experiments.

closed 2023-11-11 by seldridge