llvm/circt #4482: [FIRRTL] Use WiringProblem to handle SFC wiring transform

sam-shahrestani, 2022-12-23:
Quite a few mature codebases are still using Chisel's `BoringUtils` and the `firrtl.passes.wiring._` set of annotations to express cross-module references. 

It seems that with the introduction of WiringProblem. a reasonable way of handling these Annotations has come up. For example, I've had success tracking the defined Source/SinkAnnotations and their "pin" definitions in a custom annotation resolver and then emitting WiringProblems as needed.

The biggest question is probably if such a thing is needed at all (i.e. if it is preferable to use more specific transforms, similar to how `WireDFT` handles a specific case, or some other method of handling these has been looked at).

seldridge, 2022-12-23:
This has been a sticking point for a number of code bases migrating to CIRCT. Given that the `WiringProblem` infra is about one patch away from supporting this, I say we just do it.

If you've got something you want to PR, please go for it!

There are some changes that I've been thinking about making here that will eventually affect the `BoringUtils` API and this work. However, I'm happy to just land a replacement for `BoringUtils` as it exists today.

1. The use of two annotations (`SourceAnnotation` and `SinkAnnotation` linked by a `pin: String` key) is wrong. This is better modeled as a single, unified annotation that takes one source, one sink, and a boolean that indicates if the source -> LCA path should be bored with real or ref type ports.
2. Following from (1), all the `BoringUtils` APIs should be deprecated (`addSource`, `addSink`, and `bore`) and replaced with two APIs: (a) a replacement for `BoringUtils.bore` which takes a single source and _exactly one_ sink generating real ports and (b) `BoringUtils.tap` which has the same shape, but generates ref type ports along the source -> LCA path. (I.e., this is just a thin wrapper around the new annotation.) More complicated APIs can be built in Chisel to provide capabilities like `addSource`/`addSink` or multi-sink boring, but that emit as this simpler annotation which allows us to scale back the complexity of tracking `pin` keys or rejecting invalid `WiringProblems`.
3. Mid-term, I think we can move the resolution of this API into Chisel and do boring (real or ref type) directly and remove the `WiringProblem`.

`WireDFT` will get junked if we can replace it with this. 😉

CC: @debs-sifive (who has been poking at this, too)
CC: @oharboe (who was blocked by not having this)

sam-shahrestani, 2022-12-29:
I've cleaned up my work and opened a PR. This is a first-time contribution to CIRCT so feedback is very welcome :)

dtzSiFive, 2023-02-10:
Closing, fixed with #4496! Thanks!

closed 2023-02-10 by dtzSiFive