llvm/circt #4483: [Comb] Flow-sensitive? mux condition canonicalizer

uenoku, 2022-12-23:
Input:
```
circuit Foo:
  module Foo:
     input clock: Clock
     input a: UInt<1>
     input b: UInt<4>
     output out: UInt<4>
     reg release_state: UInt<4>, clock 
     release_state <= b
     node _T_2238 = eq(release_state, UInt<4>("h5"))
     when _T_2238 : 
       release_state <= UInt<4>("h0") 
     node _T_2240 = eq(release_state, UInt<4>("h2")) 
     when _T_2240 : 
       release_state <= UInt<4>("h7") 
     when _T_2238 :
       release_state <= UInt<4>("h2") 

     out <= release_state
```

Current output:
```
  always @(posedge clock) begin
    automatic logic _T_2238;
    _T_2238 = release_state == 4'h5; 
    if (_T_2238) 
      release_state <= 4'h2; 
    else if (release_state == 4'h2)
      release_state <= 4'h7;
    else if (_T_2238)  
      release_state <= 4'h0;
    else 
      release_state <= b;
  end // always @(posedge)
```

The second `if(_T_2238)` should be removed because that's clearly dead code. The canonicalizer can be implemented in comb mux canonilizer by analyzing mux chain's conditions.

```mlir
hw.module @Foo(%clock: i1, %a: i1, %b: i4) -> (out: i4) {
  %c5_i4 = hw.constant 5 : i4
  %c0_i4 = hw.constant 0 : i4
  %c2_i4 = hw.constant 2 : i4
  %c7_i4 = hw.constant 7 : i4
  %release_state = seq.firreg %4 clock %clock {firrtl.random_init_start = 0 : ui64} : i4
  %0 = comb.icmp bin eq %release_state, %c5_i4 {sv.namehint = "_T_2238"} : i4
  %1 = comb.mux bin %0, %c0_i4, %b : i4
  %2 = comb.icmp bin eq %release_state, %c2_i4 {sv.namehint = "_T_2240"} : i4
  %3 = comb.mux bin %2, %c7_i4, %1 : i4
  %4 = comb.mux bin %0, %c2_i4, %3 : i4
  hw.output %release_state : i4
}
```

Should be optimized into 
```
```mlir
hw.module @Foo(%clock: i1, %a: i1, %b: i4) -> (out: i4) {
  %c5_i4 = hw.constant 5 : i4
  %c0_i4 = hw.constant 0 : i4
  %c2_i4 = hw.constant 2 : i4
  %c7_i4 = hw.constant 7 : i4
  %release_state = seq.firreg %3 clock %clock {firrtl.random_init_start = 0 : ui64} : i4
  %0 = comb.icmp bin eq %release_state, %c5_i4 {sv.namehint = "_T_2238"} : i4
  %1 = comb.icmp bin eq %release_state, %c2_i4 {sv.namehint = "_T_2240"} : i4
  %2 = comb.mux bin %1, %c7_i4, %b : i4
  %3 = comb.mux bin %0, %c2_i4, %2 : i4
  hw.output %release_state : i4
}
```
```

uenoku, 2022-12-26:
Similar example:
```
circuit Foo:
  module Foo:
    input a: UInt<2>
    input b: UInt<1>
    output c: UInt<1>
    c <= mux(eq(a, UInt<2>(1)), eq(a, UInt<2>(2)), b)
```

current output:
```
  assign c = a == 2'h1 ? a == 2'h2 : b;
```
ideally we want to canonicalize a==2'h2 to false.
```
  assign c = a == 2'h1 ? 2'h0 : b;
```

fabianschuiki, 2025-07-11:
This is covered in part by #8685. (The first example is. The second isn't, since that PR doesn't really know how to handle equality checks and what they imply on other equality checks.)