llvm/circt #4494: [PyCDE] dot_prod_system.py integration test broken

teqdruid, 2022-12-28:
```
FAIL: PyCDE Integration :: pytorch/dot_prod_system.py (1 of 3)
******************** TEST 'PyCDE Integration :: pytorch/dot_prod_system.py' FAILED ********************
Script:
--
: 'RUN: at line 2';   rm -rf /home/jodemme/internal_circt_build/tools/circt/frontends/pycde/integration_test/pytorch/Output/dot_prod_system.py.tmp
: 'RUN: at line 3';   mlir-opt  /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot.linalg.mlir --one-shot-bufferize="allow-return-allocs bufferize-function-boundaries" --buffer-results-to-out-params --convert-linalg-to-affine-loops --lower-affine --convert-scf-to-cf --canonicalize > dot.cf.mlir
: 'RUN: at line 4';   hlstool dot.cf.mlir --with-esi --dynamic-hw -ir -ir-output-level 2 > dot.hw.mlir
: 'RUN: at line 5';   "/usr/bin/python3.9" /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot_prod_system.py /home/jodemme/internal_circt_build/tools/circt/frontends/pycde/integration_test/pytorch/Output/dot_prod_system.py.tmp 2>&1
: 'RUN: at line 6';   /home/jodemme/internal_circt_build/bin/esi-cosim-runner.py --no-aux-files --tmpdir /home/jodemme/internal_circt_build/tools/circt/frontends/pycde/integration_test/pytorch/Output/dot_prod_system.py.tmp --schema /home/jodemme/internal_circt_build/tools/circt/frontends/pycde/integration_test/pytorch/Output/dot_prod_system.py.tmp/runtime/schema.capnp /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot_prod_system.py `ls /home/jodemme/internal_circt_build/tools/circt/frontends/pycde/integration_test/pytorch/Output/dot_prod_system.py.tmp/hw/*.sv | grep -v driver.sv`
--
Exit Code: 1

Command Output (stderr):
--
/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot.linalg.mlir:3:8: error: 'tensor.empty' op cannot be bufferized, but can be converted to bufferization.alloc_tensor
  %0 = tensor.empty() : tensor<i32>
       ^
/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot.linalg.mlir:3:8: note: see current operation: %3 = "tensor.empty"() : () -> tensor<i32>
/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot.linalg.mlir:3:8: error: failed to bufferize op
  %0 = tensor.empty() : tensor<i32>
       ^
/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/frontends/PyCDE/integration_test/pytorch/dot.linalg.mlir:3:8: note: see current operation: %3 = "tensor.empty"() : () -> tensor<i32>

--

********************
********************
Failed Tests (1):
  PyCDE Integration :: pytorch/dot_prod_system.py
```

The `mlir-opt` command is failing due to an upstream change. Let's switch to committing the output of that command so we're _slightly_ more resilient to (irrelevant IMO) upstream changes. Do you agree @mortbopet ?

mortbopet, 2022-12-29:
I can see arguments for both ways, although I'd be in favor of fixing the internal test when this comes up. `dot_prod_system.py` is an integration test, and we should expect it to move in lockstep with the rest of CIRCTs' dependency on upstream MLIR. Suppose we have passes that ingest dialects generated by upstream passes, and CIRCT compiles for a given upstream version. In that case, our tests should likewise also work for the given upstream version if we claim that we're in sync with upstream.

teqdruid, 2022-12-29:
It serves as an integration test for PyCDE and CIRCT. If we're including stuff not in the CIRCT repo as part of the test, why not go from pytorch on down (and thus avoid including an MLIR asm file altogether)? That's not practical, in part due to differences in the MLIR versions used by pytorch-mlir and CIRCT.

In terms of practicality, the bufferization piece of MLIR seems to constantly be in flux. I'd like to avoid frequent changes there breaking CIRCT/PyCDE integration tests if they're not really relevant to anything CIRCT does. I suspect that the cf/arith/memref dialects and associated passes are somewhat some stable. As a compromise, I would _really_ prefer to not run the bufferization stuff. The rest of the passes seem like they're likely more stable.

teqdruid, 2023-01-14:
Fixed by #4531

closed 2023-01-14 by teqdruid