llvm/circt #4561: [ExportVerilog] Add Emission Option for Unique Case-insensitive Names in Module Scope

seldridge, 2023-01-20:
Some lint tools are unhappy if there are any names which conflict ignoring case. See STARC-1.1.1.5 (page 14 of https://picture.iczhiku.com/resource/eetop/WHKEgQWrYgOkYcnv.pdf). While this rule is clearly, clearly intended for human-written Verilog and not compiler-generated Verilog, it is one by default in some lint tools.

Add an emission option for `ExportVerilog`/`PrepareForEmission` that will unique names within a module ignoring case. This should include the module name in the check.

E.g., in the following, everything conflicts when ignoring case:

```mlir
module {
  hw.module @Foo(%Foo: i1) -> (foo: i1) {
    %FOO = sv.wire  : !hw.inout<i1>
    %0 = sv.read_inout %FOO : !hw.inout<i1>
    sv.assign %FOO, %Foo : i1
    hw.output %0 : i1
  }
}
```

Currently this produces:

```verilog
module Foo(
  input  Foo,
  output foo);

  wire FOO = Foo;
  assign foo = FOO;
endmodule
```

Instead, this should produce something like:

```verilog
module Foo(
  input  Foo_0,
  output foo_1);

  wire FOO_2 = Foo_0;
  assign foo_1 = FOO_2;
endmodule
```

teqdruid, 2023-01-20:
> While this rule is clearly, clearly intended for human-written Verilog and not compiler-generated Verilog

Could also be for human-read code.

seldridge, 2023-01-20:
It looks like VHDL is case-insensitive so we will need our manglers to be able to do this eventually.