llvm/circt #4565: Build of CIRCT fails due to error with string sub-command REGEX mode MATCH

vani2620, 2023-01-21:
Trying to build this project from source and I get the following error when attempting to build it. Running on Windows 64-bit, using MSVC as build generator. I've attached the CMakeError and CMakeOutput files below, and I've also copied the terminal output to a pastebin [here](https://pastebin.com/JgFQ5sp6).
[CMakeError.log](https://github.com/llvm/circt/files/10471110/CMakeError.log)
[CMakeOutput.log](https://github.com/llvm/circt/files/10471111/CMakeOutput.log)

teqdruid, 2023-01-21:
MLIR (and thus CIRCT) require a reasonably new version of cmake. What version are you using? IIRC the version which comes with VS is too old.

vani2620, 2023-01-21:
I'm using CMake version 3.25.1.

teqdruid, 2023-01-21:
That's good enough.

Seems to be in the verilator detection code. I've never compiled against verilator on Windows. What does running "C:/Users/vani2/oss-cad-suite/bin/verilator --version" from a terminal print? If it's anything other than a version number, that's the problem.

Either way, if you try with the `VERILATOR_DISABLE` cmake flag set, you should get beyond this error.

vani2620, 2023-01-21:
Yeah, it gives me an error.

> PS C:\Users\vani2> C:\Users\vani2oss-cad-suite\bin\verilator --version
> C:\Users\vani2oss-cad-suite\bin\verilator : The term 'C:\Users\vani2oss-cad-suite\bin\verilator' is not recognized as
> the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was
> included, verify that the path is correct and try again.
> At line:1 char:1
> + C:\Users\vani2oss-cad-suite\bin\verilator --version
> + ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>     + CategoryInfo          : ObjectNotFound: (C:\Users\vani2o...e\bin\verilator:String) [], CommandNotFoundException
>     + FullyQualifiedErrorId : CommandNotFoundException
 
So I just use `cmake <the usual shit> VERILATOR_DISABLE=ON` to fix it?

vani2620, 2023-01-21:
Oh, wait, I messed up the path there. Fixed it, but now it just opens the prompt that pops up when Windows comes across a file extension it doesn't recognize. Opened it with VS Code for the hell of it and it opens up a file named "verilator".

teqdruid, 2023-01-21:
Yeah, verilator on Windows is different from on Linux. It has a perl based wrapper to its main executable.

> 
> So I just use `cmake <the usual shit> VERILATOR_DISABLE=ON` to fix it?
> 

it's `-DVERILATOR_DISABLE=ON`, but yeah

vani2620, 2023-01-21:
Gotcha, thanks!

teqdruid, 2023-01-21:
lemme know if that does it for ya

vani2620, 2023-01-21:
Yeah, build files were successfully written, it seems. Thank you so much!

teqdruid, 2023-01-21:
I'm gonna re-open this as a tracking issue. We should really get this working.

teqdruid, 2023-01-21:
RCA: Verilator (both on Windows and Linux) has an executable (named verilator_bin[.exe]) which is wrapped by a perl script which calls the executable. Fine on Linux, but Windows doesn't know how to execute perl scripts and perl may not even be installed.