llvm/circt #4569: Build (apparently) fails due to issue with MooreTypes.cpp file

vani2620, 2023-01-21:
So I tried actually building the project after the build files were generated and it looks like it failed because of a reference to deleted function error? I've copied the command line output and posted it on Pastebin [here](https://pastebin.com/WZ5rqNfT).

teqdruid, 2023-01-21:
While I can't help you with this specific issue, I will note that we only test the VS2019 compiler, not 2022. I'm pretty much the only CIRCT dev here who has to care about Windows, so our Windows support is spotty at best.

teqdruid, 2023-01-21:
I can confirm that this source doesn't compile with VS2022.

teqdruid, 2023-01-21:
```
C:\code\circt\lib\Dialect\Moore\MooreTypes.cpp(1448): error C2280: 'mlir::DialectAsmPrinter::DialectAsmPrinter(const mlir::DialectAsmPrinter &)': attempting to reference a deleted function
C:\code\circt\llvm\llvm\..\mlir\include\mlir/IR/DialectImplementation.h(32): note: compiler has generated 'mlir::DialectAsmPrinter::DialectAsmPrinter' here
C:\code\circt\llvm\llvm\..\mlir\include\mlir/IR/DialectImplementation.h(32): note: 'mlir::DialectAsmPrinter::DialectAsmPrinter(const mlir::DialectAsmPrinter &)': function was implicitly deleted because a base class invokes a deleted or inaccessible function 'mlir::AsmPrinter::AsmPrinter(const mlir::AsmPrinter &)'
C:\code\circt\llvm\llvm\..\mlir\include\mlir/IR/OpImplementation.h(221): note: 'mlir::AsmPrinter::AsmPrinter(const mlir::AsmPrinter &)': function was explicitly deleted
C:\code\circt\llvm\llvm\include\llvm/ADT/TypeSwitch.h(122): note: see reference to function template instantiation 'mlir::DialectAsmPrinter customTypePrinter::<lambda_2>::operator ()<U>(_T1) const' being compiled
```

For some reason VS2022 didn't like `return printer << "void", success();` in a lambda.

@vani2620 I spoke too soon -- turns out I could help. The latest main fixes the build, but we have some failing tests with VS2022. *sigh*

closed 2023-01-21 by teqdruid

vani2620, 2023-01-21:
Latest main...? As in, I should re-clone the repository and all?

I'd have loved to do this on Linux, but I'd really rather not set up a dual boot and WSL2 is a pain in the ass and refuses to work no matter what I try for whatever godforsaken reason.

teqdruid, 2023-01-21:
Just `git pull` in the main branch should do it.

WSL2 is pretty good once it's up and running. It's just a memory hog since
WSL2 runs a VM in the background.

vani2620, 2023-01-21:
I figured out how to manually edit in the changes just as you posted this, haha. But for future reference, the main branch would've been the circt folder, correct?

commit 470153b9 (2023-01-21): [Moore] Fix Visual Studio 2022 compilation error

Fixes #4569. For some reason VS2022 doesn't like the shorthand that was
being used here. This fixes it.
commit ea4acd3f (2023-01-21): [Moore] Fix Visual Studio 2022 compilation error

Fixes #4569. For some reason VS2022 doesn't like the shorthand that was
being used here. This fixes it.