llvm/circt #4570: [Calyx] Tests fail in VS2022 compile

teqdruid, 2023-01-21:
```
Failed Tests (6):
  CIRCT :: Conversion/PipelineToCalyx/convert_pipeline.mlir
  CIRCT :: Conversion/SCFToCalyx/cider_source_location.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_simple.mlir
  CIRCT :: Conversion/SCFToCalyx/errors.mlir
```

We don't have a VS2022 CI flow, but when I run a compile locally (after fixing #4569), I get these test failures. Full log:

```
-- Testing: 496 tests, 1 workers --
Testing:
FAIL: CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir (2 of 496)
******************** TEST 'CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   c:\code\circt_build_2022\bin\circt-opt.exe C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir --lower-scf-to-calyx -canonicalize -split-input-file | c:\code\circt_build_2022\bin\filecheck.exe C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir
--
Exit Code: 1

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "c:\code\circt_build_2022\bin\circt-opt.exe" "C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir" "--lower-scf-to-calyx" "-canonicalize" "-split-input-file"
note: command had no output on stdout or stderr
error: command failed with exit status: 1
$ "c:\code\circt_build_2022\bin\filecheck.exe" "C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir"
# command stderr:
C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir:65:17: error: CHECK-LABEL: expected string not found in input
// CHECK-LABEL: calyx.component @main(%in0: i32, %in1: i32, %in2: i32, %clk: i1 {clk}, %reset: i1 {reset}, %go: i1 {go}) -> (%out0: i32, %done: i1 {done}) {
                ^
<stdin>:1:1: note: scanning from here

^

Input file: <stdin>
Check file: C:\code\circt\test\Conversion\SCFToCalyx\convert_controlflow.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
          1:
label:65     X error: no match found
          2: // -----
label:65     ~~~~~~~~~
          3:
label:65     ~
          4: // -----
label:65     ~~~~~~~~~
          5:
label:65     ~
          6: // -----
label:65     ~~~~~~~~~
>>>>>>

error: command failed with exit status: 1

--

********************
Testing:
FAIL: CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir (3 of 496)
******************** TEST 'CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   c:\code\circt_build_2022\bin\circt-opt.exe C:\code\circt\test\Conversion\SCFToCalyx\convert_memory.mlir --lower-scf-to-calyx -canonicalize -split-input-file | c:\code\circt_build_2022\bin\filecheck.exe C:\code\circt\test\Conversion\SCFToCalyx\convert_memory.mlir
[... 178 lines cut ...]
Script:
--
: 'RUN: at line 1';   c:\code\circt_build_2022\bin\circt-opt.exe --lower-scf-to-calyx C:\code\circt\test\Conversion\SCFToCalyx\errors.mlir -split-input-file -verify-diagnostics
--
Exit Code: 1

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "c:\code\circt_build_2022\bin\circt-opt.exe" "--lower-scf-to-calyx" "C:\code\circt\test\Conversion\SCFToCalyx\errors.mlir" "-split-input-file" "-verify-diagnostics"
# command output:

// -----

// -----

# command stderr:
within split at C:\code\circt\test\Conversion\SCFToCalyx\errors.mlir:23 offset :8:6: error: expected error "CFG backedge detected. Loops must be raised to 'scf.while' or 'scf.for' operations." was not produced
  // expected-error @+1 {{CFG backedge detected. Loops must be raised to 'scf.while' or 'scf.for' operations.}}
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error: command failed with exit status: 1

--
```

dtzSiFive, 2023-01-22:
Not sure re:`errors` test, but FWIW those failing tests have been triggering ASAN, which reports heap-use-after-free (#4378).

It's know we're doing something re:rewriterpatterns so perhaps that's more problematic now (#4381) .

rachitnigam, 2023-01-28:
Did https://github.com/llvm/circt/pull/4592 fix this issue?

mikeurbach, 2023-01-31:
Yep

closed 2023-01-31 by mikeurbach