llvm/circt #4584: [FIRRTL] Raw Param Difference vs. SFC

seldridge, 2023-01-24:
Currently, CIRCT will squash "raw" FIRRTL external module parameters into string parameters. This produces a difference with the SFC. Consider:

```
circuit Foo:
  extmodule Bar:
    output a: UInt<1>
    parameter foo = 'hello'

  module Foo:
    output a: UInt<1>

    inst bar of Bar
    a <= bar.a
```

This compiles with CIRCT to:

```veriloog
module Foo(
  output a);

  Bar #(
    .foo("hello")
  ) bar (
    .a (a)
  );
endmodule
```

With SFC, this produces:

```verilog
module Foo(
  output  a
);
  wire  bar_a;
  Bar #(.foo(hello)) bar (
    .a(bar_a)
  );
  assign a = bar_a;
endmodule
```

youngar, 2023-01-24:
In the output verilog, what is `hello`?  Is it an identifier (or possibly macro) defined by hand in Verilog compilation unit?
If that is what it is, it seems pretty dangerous, like we might need external global values defined in FIRRTL. I know we're far away from anything like type-checking parameters, but example of what this might look like:
```firrtl
circuit Foo:
  extvalue hello : String
  extmodule Bar:
    parameter foo : String = hello  
```

seldridge, 2023-01-25:
It's likely some macro.

I think a better name for this than "raw param" would be "verbatim param". This is outside of the bounds of anything that we could check (which is the intention).

A likely better alternative would be to reject this in the parser. These raw params aren't in the spec (though integer and string parameters aren't mentioned other than by example). We do have one usage of this internally (though I have no idea if it is load bearing) and I see some usage in the open source.

seldridge, 2023-01-25:
@jackkoenig or @azidar: you two have any thoughts about raw param support?

jackkoenig, 2023-01-25:
They are potentially useful for some SystemVerilog blackboxes. They can be used to do a shitty version of parametric polymorphism (sort of like what Go did in their standard library up until they added generics last year). It's not a design pattern I would endorse (just use Chisel 😉), but flat out banning them in FIRRTL prevents Chisel/FIRRTL users from instantiating SystemVerilog modules that do use this. I'm generally of the opinion of trying to be maximally compatible when it comes to interfacing with blackboxes, and I don't think there's a high cost of support here, but I may be wrong.

darthscsi, 2023-01-25:
verbatim is single quotes?  I'm quite curious what kind of escaping goes on...

PR #5774 (2023-08-08): [FIRRTL][HW] Add verbatim literal support
Currently, FIRRTL params values wrapped in 'single-quotes' are emitted as string in the verilog output, but they should be emitted verbatim.

This PR corrects this behaviour in the FIRParser, by emitting hw::ParamVerbatimAttr instead of StringAttr for raw/verbatim string params.

Summary of changes:

- Rename "raw string" to "verbatim string" in the FIRRTL parser and lexer.
- Add new builder for ParamVerbatimAttr that sets the type to NoneType.
- Update the FIRRTL parser to emit a hw::ParamVerbatimAttr of type NoneType.

Fixes: #4584
commit f3e4bc69 (2023-08-08): [FIRRTL][HW] Add verbatim literal support (#5774)

Currently, FIRRTL params values wrapped in 'single-quotes' are emitted as
string in the verilog output, but they should be emitted verbatim.

This PR corrects this behaviour in the FIRParser, by emitting
hw::ParamVerbatimAttr instead of StringAttr for raw/verbatim string params.

Summary of changes:

- Rename "raw string" to "verbatim string" in the FIRRTL parser and lexer.
- Add new builder for ParamVerbatimAttr that sets the type to NoneType.
- Update the FIRRTL parser to emit a hw::ParamVerbatimAttr of type NoneType.

Fixes: #4584
closed 2023-08-08 by rwy7