llvm/circt #4588: [FIRRTL] Support read-under-write modifiers on CHIRRTL mem constructs

seldridge, 2023-01-25:
CHIRRTL memories were supposed to have an `undefined`, `old`, or `new` read-under-write modifiers after the memory declaration. E.g.:

```
smem mem1 : UInt<8> [8] old
smem mem2 : UInt<8> [8] new
smem mem3 : UInt<8> [8] undefined
```

This then should be copied over to the FIRRTL memory that is inferred.

Presently, this modifier was never serialized in the textual format coming out of Chisel. (It was just dropped.)

This will change in the actual release of Chisel 3.6.

mwachs5, 2023-01-25:
when you say "was supposed to" is there something in the firrtl spec that says this is supposed to be there? And is the default actually undefined according to the firrtl spec?

mwachs5, 2023-01-25:
my read of the spec is no default is allowed, you have to specify something. https://github.com/chipsalliance/firrtl-spec/blob/0190aba6a059c885eff2a01765b3966e81db4533/spec.md#read-under-write-behavior

seldridge, 2023-01-25:
(Likely answered elsewhere and documented here for completeness:) These are the CHIRRTL constructs which are undocumented in the spec:

- `smem` -- a synchronous read memory)
- `cmem` -- a combinational read memory)
- `( read | write ) mport` -- a read or write port of one of the above

This was never documented in the FIRRTL spec. This hasn't happened because no-one was happy with the current state of how these work. They rely on heuristics to infer the read enable and the clock and the way that Chisel emits them violates expected scoping rules.

By "supposed to", there was a change in 2020 that added read-under-write modifiers to both `cmem` and `smem`. This never updated the FIRRTL spec (as there was no place to update it!). These were made to work with Chisel and FIRRTL, but the _serialization_ to FIRRTL IR w/ CHIRRTL extensions was never done. The read-under-write behavior was only preserved if passed in an in-memory representation to a FIRRTL compiler. This was only recently realized when porting Chisel tests to use CIRCT where the read-under-write behavior was not `undefined` or `new` and the FIRRTL/CHIRRTL text was serialized to a file which dropped the read-under-write information.

programmerjake, 2024-12-06:
circt also doesn't respect read-under-write for firrtl memories (which are part of the firrtl spec): #7957