llvm/circt #4606: [CalyxToFSM] Add `par` FSM lowering

mortbopet, 2023-01-31:
[`CalyxToFSM`](https://github.com/llvm/circt/blob/main/lib/Conversion/CalyxToFSM/CalyxToFSM.cpp#L43) is currently missing support for lowering the `calyx.par` control operation to an FSM construct. This is an important construct to support seeing as most SCFToCalyx designs contain it.

For inspiration, [the native compilers' implementation of control compilation of the `par` construct](https://docs.calyxir.org/source/src/calyx/passes/top_down_compile_control.rs.html#903-990).

rachitnigam, 2023-08-22:
One workaround for this is using the new `calyx-native` pass (#5879) which can lower `par` and generate an MLIR representation for consumption in future CIRCT passes. Again, it would be nice to have this implemented within CIRCT but just documenting this workaround for now.

mortbopet, 2023-08-22:
It would indeed be nice - I _think_ par lowering is the last "big" missing feature before most (all?) Calyx programs can be compiled entirely in CIRCT.

rachitnigam, 2023-08-22:
Right! Although there is always an ever-growing list of new feature in the native language being added as we build out the new HLS flow.