llvm/circt #4614: Integration test failing

youngar, 2023-02-02:
The nightly integration test is failing in AffineToPipeline.  I am guessing this is due to a recent LLVM bump.

https://github.com/llvm/circt/actions/runs/4074234845/jobs/7019130474#step:7:4682

```
circt/test/Conversion/AffineToPipeline/loops.mlir:106:12: error: CHECK: expected string not found in input
 // CHECK: pipeline.while II = 3 
           ^   
<stdin>:88:31: note: scanning from here
 func.func @dot_mul_accumulate(%arg0: memref<64xi32>, %arg1: memref<64xi32>) -> i32 {
                              ^
<stdin>:93:7: note: possible intended match here
 %0 = pipeline.while II = 1 trip_count = 64 iter_args(%arg2 = %c0, %arg3 = %c0_i32) : (index, i32) -> i32 {
      ^   

Input file: <stdin>
Check file: circt/test/Conversion/AffineToPipeline/loops.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
             .   
             .   
             .   
            83:  } : i32, i32, index·
            84:  pipeline.terminator iter_args(%3#2, %3#1), results(%3#1) : (index, i32) -> i32·
            85:  }·  
            86:  return %2 : i32·
            87:  }·  
            88:  func.func @dot_mul_accumulate(%arg0: memref<64xi32>, %arg1: memref<64xi32>) -> i32 {·
check:106'0                                   X~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ error: no match found
            89:  %c0_i32 = arith.constant 0 : i32·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            90:  %c0 = arith.constant 0 : index·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            91:  %c64 = arith.constant 64 : index·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            92:  %c1 = arith.constant 1 : index·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            93:  %0 = pipeline.while II = 1 trip_count = 64 iter_args(%arg2 = %c0, %arg3 = %c0_i32) : (index, i32) -> i32 {·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
check:106'1           ?                                                                                                      possible intended match
            94:  %1 = arith.cmpi ult, %arg2, %c64 : index·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            95:  pipeline.register %1 : i1· 
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~
            96:  } do {·
check:106'0     ~~~~~~~~
            97:  %1:3 = pipeline.while.stage start = 0 {·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            98:  %4 = memref.load %arg0[%arg2] : memref<64xi32>·
check:106'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

mikeurbach, 2023-02-02:
Hmm, I would have expected this test (in `circt/test`) to be covered by CI on PR / merge to main. Strange that the nightly integration failed. Is it possible this test is non-determinate @matth2k? I can help take a look at it.

mikeurbach, 2023-02-02:
Ah I think this is our issue: https://github.com/llvm/circt/blob/a249d6e1df57eadb985811da4f35016c6fe2520d/lib/Conversion/AffineToPipeline/AffineToPipeline.cpp#L88

It's not obvious, but there should be no side effects in an `assert`. `assert` is a macro, and only builds with asserts enabled will execute that code. That would explain why check-circt passes in PR/main CI, but shows up in integration test, where we have exercise more corners of the build matrix.

If this is indeed the issue, I'll push up a quick fix.

closed 2023-02-02 by mikeurbach