llvm/circt #4638: [FIRRTL][HW?] Port Simplification Pass

seldridge, 2023-02-07:
For certain designs within a separable compilation unit (without a fixed port-level interface), it can be beneficial to "simplify" the connections between them by removing ports when possible. I think this can be pretty easily handled by allowing for any "U-turn" connections to be removed.  

E.g., consider the following. This is conceptually doing `qux.a <= baz.a`. However, it is being done in a roundabout way.

```
circuit Foo:
  extmodule Baz:
    output a: UInt<1>

  extmodule Qux:
    input a: UInt<1>

  module Bar:
    output baz_a: UInt<1>
    input qux_a: UInt<1>

    inst baz of Baz
    inst qux of Qux
    baz_a <= baz.a
    qux.a <= qux_a

  module Foo:

    inst bar of Bar
    bar.qux_a <= bar.baz_a
```

CIRCT currently preserves the port-level interface to produce:

```verilog
// external module Baz

// external module Qux

module Bar(
  input  qux_a,
  output baz_a
);

  Baz baz (
    .a (baz_a)
  );
  Qux qux (
    .a (qux_a)
  );
endmodule

module Foo();
  wire _bar_baz_a;
  Bar bar (
    .qux_a (_bar_baz_a),
    .baz_a (_bar_baz_a)
  );
endmodule
```

However, this could be simplified to:

```verilog
// external module Baz

// external module Qux

module Bar();
  wire x;
  Baz baz (
    .a (x)
  );
  Qux qux (
    .a (x)
  );
endmodule

module Foo();
  Bar bar ();
endmodule
```

Roughly the cost model is "a port costs 1 unit" and no port is free. There is a similar approach that can be taken to convert real ports to ref type ports.

This may be beneficial to have on HW as opposed to just on FIRRTL.

This approach has problems because it fundamentally destroys design intent. It's also sketchy to rely on this for anything other than optimization within a hierarchy whose internals you do not care about.

dtzSiFive, 2023-02-08:
Awesome! Been wondering about this for a while now, too.

1) As for being problematic/against design intent, I think this would pair well with mechanism(s) for users to specify their design intent, aka the internals they DO care about.  For example, a wire that looks strange and routed unnecessarily "costly" but is done that way intentionally for layout reasons (ideally using something more nuanced than our "DontTouch" hammer).  And we already want interfaces "that matter" to be marked as such somehow.  This seems useful regardless re:optimizing (IMCP, IMDCE, etc.).

2) What's the connection to ref type ports/what do you mean there, would you mind elaborating a bit?  Thanks!

Re-optimizing u-turns, I think you're suggesting:
Iteratively replacing "u-turns"  (sink flow port/field is necessarily driven entirely by source flow port/field, or so, to start with) with direct (equivalent) connections in the parent/child (hoisting/sinking the connection through the ports + dropping or letting IMDCE kill the now-dead ports), is that about right?

It'd be neat to "eventually" optimize context-sensitively (k-limited?), so like in that example if there is more than one instance of Bar but all have their ports tied together similarly, we could do the same optimization re:Bar and its ports.  But we don't do this "yet" elsewhere so not suggesting it's a limitation as much as wondering if we can frame this to get that without being too expensive :smile: .  Anyway...

seldridge, 2023-02-08:
> What's the connection to ref type ports/what do you mean there, would you mind elaborating a bit? Thanks!

For ref types there's a slightly different local rule which may prove useful. If a reference port's user is only an input instance port, then convert the instance port to a reference type port and move the ref resolve into the instance module. There's a rough cost model here of: ports cost 2 and reference ports cost 1 (and no ports cost zero).

> Re-optimizing u-turns, I think you're suggesting:
Iteratively replacing "u-turns" (sink flow port/field is necessarily driven entirely by source flow port/field, or so, to start with) with direct (equivalent) connections in the parent/child (hoisting/sinking the connection through the ports + dropping or letting IMDCE kill the now-dead ports), is that about right?

Yep!

I suppose this is just a generalization of IMCP to propagating values. Your prior thoughts on context-sensitive IMCP likely apply here, too.