llvm/circt #4669: Binary Release Issues

dtzSiFive, 2023-02-17:
* OSX binaries dynamically link against `libzstd` which is not safe to assume exists on user's machines (this has caused multiple reports of issues from users).
* Binary tarballs should have a directory with `bin` and `lib` included.
* Probably should use the install target, we don't ship the requisite runtime library for LLHD .

cc https://github.com/llvm/circt/issues/4473 .

I don't see a good way to request zstd be linked statically, I don't think we or anything we rely on in MLIR/LLVM requires this so probably best to just disable it (`LLVM_ENABLE_ZSTD=OFF`).

jackkoenig, 2023-06-23:
I'm pretty sure all of these are fixed.

We do have `-DLLVM_ENABLE_ZSTD=OFF`, the binary tarballs are of the style `firtool-<version>/bin/firtool` now (there is no `lib` directory because the current installations are just firtool and nothing else). And that release of just firtool was achieved by switching to the install target.

closed 2023-06-23 by jackkoenig

dtzSiFive, 2023-06-25:
Great, thanks!