llvm/circt #4677: [FIRRTL][HW] Explore Register Pruning

seldridge, 2023-02-18:
The different aggregate preservation modes in `firtool` will result in different total amounts of state in the final design. It would be good to explore ways to make the total amount of state the same regardless of aggregate preservation mode.

Consider:

```
circuit Foo:
  module Foo:
    input clock: Clock
    input a: UInt<1>[2]
    output b: UInt<1>[3]

    reg r: UInt<1>[3], clock

    r[0] <= a[0]
    r[1] is invalid
    r[2] <= a[1]
    b <= r
```

Without aggregate preservation (`firtool -disable-all-randomization`) this produces:

```verilog
// Generated by CIRCT unknown git version
module Foo(
  input  clock,
         a_0,
         a_1,
  output b_0,
         b_1,
         b_2
);

  reg r_0;
  reg r_2;
  always @(posedge clock) begin
    r_0 <= a_0;
    r_2 <= a_1;
  end // always @(posedge)
  assign b_0 = r_0;
  assign b_1 = 1'h0;
  assign b_2 = r_2;
endmodule
```

With aggregate preservation (`firtool -disable-all-randomization -preserve-aggregate=all`):

```verilog
// Generated by CIRCT unknown git version
module Foo(
  input  clock,
         a_0,
         a_1,
  output b_0,
         b_1,
         b_2
);

  reg [2:0] r;
  always @(posedge clock) begin
    r[2'h0] <= a_0;
    r[2'h1] <= 1'h0;
    r[2'h2] <= a_1;
  end // always @(posedge)
  assign b_0 = r[2'h0];
  assign b_1 = r[2'h1];
  assign b_2 = r[2'h2];
endmodule
```

The ideal output is likely something like:

```verilog
module Foo(
  input  clock,
         a_0,
         a_1,
  output b_0,
         b_1,
         b_2
);

  reg [1:0] r;
  always @(posedge clock) begin
    r[2'h0] <= a_0;
    r[2'h1] <= a_1;
  end // always @(posedge)
  assign b_0 = r[2'h0];
  assign b_1 = 1'h0;
  assign b_2 = r[2'h1];
endmodule
```

mwachs5, 2023-02-18:
It's legal verilog (I thought) to have indices that aren't zero-based... keeping groups that preserve the original numbering seems less maddening potentially...

seldridge, 2023-02-18:
Hey, great suggestion, Megan. There's a couple of ways to slice this and having multiple registers is one of them. It didn't occur to me to use non-zero based register declarations. 👍

mwachs5, 2023-02-18:
I'm confused... do you want all the slices on the same power domain or different ones...?

uenoku, 2023-02-21:
One concern of removing subelemnt is that LEC would be very sensitive about the register renaming.