llvm/circt #4719: Failed hlstool with specific function.

HahaLan97, 2023-02-24:
Hi! I'm playing around with the hlstool within this project. But I got a weird error.

I'm trying to HLS a Gaussian Filter function within CIRCT.

```
#include <ctype.h>
#include <math.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#define COLS      20
#define ROWS      20
#define KERN_SIZE 3
#define KERN_RAD  (KERN_SIZE >> 1)
#define PIXELS    COLS* ROWS
#define MAX(x, y) (((x) > (y)) ? (x) : (y))
#define MIN(x, y) (((x) < (y)) ? (x) : (y))

void gauss(uint8_t input[PIXELS], uint8_t output[PIXELS])
{
    // Gauss kernel
    int16_t kernel[KERN_SIZE * KERN_SIZE] = {1, 2, 1, 2, 4, 2, 1, 2, 1};

    for (int32_t y = 0; y < ROWS; y++) {
        for (int32_t x = 0; x < COLS; x++) {
            // temp var to store the result
            int16_t sum = 0;
            for (int32_t i = -KERN_RAD; i <= KERN_RAD; i++) {
                for (int32_t j = -KERN_RAD; j <= KERN_RAD; j++) {
                    int32_t yi = y + i;
                    int32_t xj = x + j;
                    if (xj >= 0 && xj < COLS && yi >= 0 && yi < ROWS)
                        sum +=
                            (int16_t)(input[yi * COLS + xj] * kernel[(i + KERN_RAD) * KERN_SIZE + j + KERN_RAD]);
                }
            }
            // Normalize and store
            output[y * COLS + x] = (uint8_t)(MAX(MIN(255, sum >> 4), 0));
        }
    }
}
```

I firstly use `Polygeist` to translate the above C code into MLIR as following:

```
module {
  func.func @gauss(%arg0: memref<400xi8>, %arg1: memref<400xi8>) {
    %c2 = arith.constant 2 : index
    %c-1 = arith.constant -1 : index
    %c20 = arith.constant 20 : index
    %c0 = arith.constant 0 : index
    %c1 = arith.constant 1 : index
    %true = arith.constant true
    %c4_i32 = arith.constant 4 : i32
    %c255_i32 = arith.constant 255 : i32
    %c1_i32 = arith.constant 1 : i32
    %c3_i32 = arith.constant 3 : i32
    %c0_i16 = arith.constant 0 : i16
    %c20_i32 = arith.constant 20 : i32
    %c0_i32 = arith.constant 0 : i32
    %c4_i16 = arith.constant 4 : i16
    %c2_i16 = arith.constant 2 : i16
    %c1_i16 = arith.constant 1 : i16
    %alloca = memref.alloca() : memref<9xi16>
    affine.store %c1_i16, %alloca[0] : memref<9xi16>
    affine.store %c2_i16, %alloca[1] : memref<9xi16>
    affine.store %c1_i16, %alloca[2] : memref<9xi16>
[... 109 lines cut ...]
#16 0x00005600d428b005 mlir::applyPatternsAndFoldGreedily(mlir::Operation*, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) /home/bi/Desktop/circt/llvm/mlir/include/mlir/Transforms/GreedyPatternRewriteDriver.h:86:10
#17 0x00005600d4288cbc (anonymous namespace)::Canonicalizer::runOnOperation() /home/bi/Desktop/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:61:9
#18 0x00005600d481a99a mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:471:21
#19 0x00005600d481afb4 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:534:16
#20 0x00005600d4820598 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:754:36
#21 0x00005600d4820209 mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&>(mlir::MLIRContext*, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&) /home/bi/Desktop/circt/llvm/mlir/include/mlir/IR/Threading.h:46:18
#22 0x00005600d481c293 mlir::LogicalResult mlir::failableParallelForEach<std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&>(mlir::MLIRContext*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&) /home/bi/Desktop/circt/llvm/mlir/include/mlir/IR/Threading.h:92:10
#23 0x00005600d481bb37 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:764:14
#24 0x00005600d481ac67 mlir::detail::OpToOpPassAdaptor::runOnOperation(bool) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:655:5
#25 0x00005600d481a98b mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:468:5
#26 0x00005600d481afb4 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:534:16
#27 0x00005600d481c9dc mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:843:10
#28 0x00005600d481c8f2 mlir::PassManager::run(mlir::Operation*) /home/bi/Desktop/circt/llvm/mlir/lib/Pass/Pass.cpp:823:60
#29 0x00005600d3354107 doHLSFlowDynamic(mlir::PassManager&, mlir::ModuleOp, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile>>>&) /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:385:17
#30 0x00005600d3353bce processBuffer(mlir::MLIRContext&, mlir::TimingScope&, llvm::SourceMgr&, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile>>>&) /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:427:16
#31 0x00005600d3353826 processInputSplit(mlir::MLIRContext&, mlir::TimingScope&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile>>>&) /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:449:12
#32 0x00005600d335369d processInput(mlir::MLIRContext&, mlir::TimingScope&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, std::optional<std::unique_ptr<llvm::ToolOutputFile, std::default_delete<llvm::ToolOutputFile>>>&) /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:465:12
#33 0x00005600d3353554 executeHlstool(mlir::MLIRContext&) /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:500:14
#34 0x00005600d3353356 main /home/bi/Desktop/circt/tools/hlstool/hlstool.cpp:556:17
#35 0x00007f5a4e169083 __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:342:3
#36 0x00005600d335313e _start (/home/bi/Desktop/circt/build/bin/hlstool+0xe1d13e)
[1]    5147 abort (core dumped)  /home/bi/Desktop/circt/build/bin/hlstool gauss.mlir -dynamic-hw
```

Which I've asked about once on discourse, this should be a bug somewhere in the lowering.

youngar, 2023-02-24:
Crash is probably in this part of the code: https://github.com/llvm/circt/blob/55e2230e3fd12d9ce43d825fe5242a2a3899e9c3/lib/Dialect/Handshake/HandshakeOps.cpp#L341-L344

It is super weird to have the call to `rewriter.replaceOp` wrapped in `rewriter.updateRootInPlace`, and even weirder that they are changing a different operation then the one we are supposedly transacting on.  The whole thing can probably be replaced with just:
```c++
rewriter.replaceOp(op, firstParentCBranch.getDataOperand()); 
```

youngar, 2023-02-24:
Looks like the issue was actually fixed recently: https://github.com/llvm/circt/commit/083de301f0affa73d2c7f59ea73dbd03acd57bb6

HahaLan97, 2023-02-24:
> Looks like the issue was actually fixed recently: [083de30](https://github.com/llvm/circt/commit/083de301f0affa73d2c7f59ea73dbd03acd57bb6)

Oh! I'll try this later!

HahaLan97, 2023-02-28:
> Looks like the issue was actually fixed recently: [083de30](https://github.com/llvm/circt/commit/083de301f0affa73d2c7f59ea73dbd03acd57bb6)

I've pulled and rebuilt the project. This problem is already fixed! Thanks for pointing out, I'll close this issue:)

closed 2023-02-28 by HahaLan97