llvm/circt #4740: LLVM is not cached on GCC debug build

uenoku, 2023-02-28:
https://github.com/llvm/circt/actions/runs/4289964745/jobs/7473505867

teqdruid, 2023-02-28:
We probably build too much of LLVM. The regular builds are building 4250 files whereas the unified builds build 3566. If we want to keep building and caching LLVM separately, someone needs to identify all the necessary targets and compile only them.

Or we could add another corner from the integration tests (unified build) which covers GCC debug builds.

uenoku, 2023-03-01:
I agree unified build would reduce the size but is it possible to cache only llvm libs in unified build?  

> Or we could add another corner from the integration tests (unified build) which covers GCC debug builds.

+1, similarly I'm wondering we can just move gcc+debug to nightly-integration tests and test gcc+release.

teqdruid, 2023-03-01:
We want to have both a debug build and release build in the PR gates. Perhaps we should switch the order. Clang debug and gcc release?

We test a bunch of corners in the nightly, including gcc+debug. They're just all unified and use ccache instead of building all of mlir and caching all of it.

uenoku, 2023-03-01:
> Perhaps we should switch the order. Clang debug and gcc release?

Sounds good idea. Testing here: https://github.com/llvm/circt/actions/runs/4306635035

uenoku, 2023-03-02:
It passed 👍

teqdruid, 2023-03-02:
Great! Merge it!

closed 2023-03-02 by uenoku