llvm/circt #4745: CIRCT integration tests failed

Discreater, 2023-03-01:
At first, I wanted to install firtool in WSL (Arch). I install it from [AUR](https://aur.archlinux.org/packages/firtool). It can build successfully, but integration tests will fail. So, I try to follow the README and build the master branch of circt manually. And it's integration tests also failed.

Below is the test results (I use the RELEASE build type, because my machine may not have enough memory to build it in DEBUG. Other config is same to the README):
```sh
❯ ninja check-circt-integration
[0/1] Running the CIRCT integration tests
FAIL: CIRCT :: Dialect/FSM/variable/top.mlir (66 of 71)
******************** TEST 'CIRCT :: Dialect/FSM/variable/top.mlir' FAILED ********************
Script:
--
: 'RUN: at line 2';   /home/alkaid/source/circt/build/bin/circt-opt /home/alkaid/source/circt/integration_test/Dialect/FSM/variable/top.mlir --convert-fsm-to-sv --canonicalize --lower-seq-to-sv --export-split-verilog -o /home/alkaid/source/circt/build/integration_test/Dialect/FSM/variable/Output/top.mlir.tmp2.mlir
: 'RUN: at line 3';   /home/alkaid/source/circt/build/bin/circt-rtl-sim.py --compileargs="-I/home/alkaid/source/circt/build/integration_test/Dialect/FSM/variable/Output/.." top.sv /home/alkaid/source/circt/integration_test/Dialect/FSM/variable/driver.cpp --no-default-driver | /home/alkaid/source/circt/llvm/build/bin/FileCheck /home/alkaid/source/circt/integration_test/Dialect/FSM/variable/top.mlir
--
Exit Code: 1

Command Output (stderr):
--
/home/alkaid/source/circt/integration_test/Dialect/FSM/variable/top.mlir:9:16: error: CHECK-NEXT: is not on the line after the previous match
// CHECK-NEXT: out: 0
               ^
<stdin>:20:1: note: 'next' match was here
out: 0
^
<stdin>:14:7: note: previous match ended here
out: 0
      ^
<stdin>:15:1: note: non-matching line after previous match is here
out: 1
^

Input file: <stdin>
Check file: /home/alkaid/source/circt/integration_test/Dialect/FSM/variable/top.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
        .
        .
        .
       15: out: 1
       16: out: 2
       17: out: 3
       18: out: 4
       19: out: 5
       20: out: 0
next:9     !~~~~~  error: match on wrong line
       21: out: 0
>>>>>>

--

********************
********************
Failed Tests (1):
  CIRCT :: Dialect/FSM/variable/top.mlir

Testing Time: 3.37s
  Unsupported: 38
  Passed     : 32
  Failed     :  1
FAILED: integration_test/CMakeFiles/check-circt-integration /home/alkaid/source/circt/build/integration_test/CMakeFiles/check-circt-integration
cd /home/alkaid/source/circt/build/integration_test && /usr/sbin/python3.10 /home/alkaid/source/circt/llvm/build/./bin/llvm-lit -sv /home/alkaid/source/circt/build/integration_test
ninja: build stopped: subcommand failed.
```

uenoku, 2023-03-01:
This is tested in our integeration test CI so maybe this is a bug of simulator. Could you tell me which version of verilator are you using?

Discreater, 2023-03-01:
Verilator 5.006 2023-01-22 rev UNKNOWN.REV

uenoku, 2023-03-01:
Thanks, I was able to reproduce verilator 5.004. I also tested verilator 4.224 and it passes. Presumably it's a bug of verilator.

uenoku, 2023-03-07:
Interesting, @mortbopet could you check which is correct behavior?

mortbopet, 2023-03-10:
Hmm... I do not seem to be able to reproduce the error - versions 5.004, 5.007, 4.225 all yield the same outputs on my end.

mortbopet, 2023-09-29:
Given the multiple reports of failures on this test, and the fact that the FSM dialect isn't load bearing for any significant work at the moment, I'll disable this test for now. I hopefully will get back into the FSM dialect at some point in the future and do a deeper dive.

closed 2023-09-29 by mortbopet