llvm/circt #4748: Dialect `firrtl.module` is missing a `name` attribute?

SpriteOvO, 2023-03-02:
[The doc of `firrtl.module`](https://circt.llvm.org/docs/Dialects/FIRRTL/#firrtlmodule-circtfirrtlfmoduleop) described only 1 attr `annotations`.

This is what I tried:

```c
#include <stdio.h>

#include <circt-c/Dialect/FIRRTL.h>
#include <circt-c/ExportFIRRTL.h>

#define MK_STR(s) (mlirStringRefCreateFromCString((s)))
#define ARRAY_SIZE(arr) (sizeof(arr) / sizeof((arr)[0]))

void export(MlirStringRef export, void *userData)
{
  printf("%.*s", export.length, export.data);
}

int main(void)
{
  MlirContext ctx = mlirContextCreate();
  MlirDialectHandle firrtlDialect = mlirGetDialectHandle__firrtl__();
  mlirDialectHandleLoadDialect(firrtlDialect, ctx);

  MlirLocation unkLoc = mlirLocationUnknownGet(ctx);

  MlirModule module = mlirModuleCreateEmpty(unkLoc);
  MlirBlock body = mlirModuleGetBody(module);

  MlirOperationState circuitState = mlirOperationStateGet(MK_STR("firrtl.circuit"), unkLoc);
  MlirNamedAttribute circuitAttrs[] = {
      mlirNamedAttributeGet(
          mlirIdentifierGet(ctx, MK_STR("name")),
          mlirAttributeParseGet(ctx, MK_STR("\"TestCir\""))),
      mlirNamedAttributeGet(
          mlirIdentifierGet(ctx, MK_STR("annotations")),
          mlirAttributeParseGet(ctx, MK_STR("[]"))),
  };
  mlirOperationStateAddAttributes(&circuitState, ARRAY_SIZE(circuitAttrs), circuitAttrs);

  MlirRegion circuitRegion = mlirRegionCreate();
  MlirOperationState moduleState = mlirOperationStateGet(MK_STR("firrtl.module"), unkLoc);

  MlirNamedAttribute moduleAttrs[] = {
      mlirNamedAttributeGet(
          mlirIdentifierGet(ctx, MK_STR("annotations")),
          mlirAttributeParseGet(ctx, MK_STR("[]"))),
  };
  mlirOperationStateAddAttributes(&moduleState, ARRAY_SIZE(moduleAttrs), moduleAttrs);
  MlirOperation moduleOp = mlirOperationCreate(&moduleState);
  MlirBlock circuitBlock = mlirBlockCreate(0, NULL, NULL);

  mlirBlockAppendOwnedOperation(circuitBlock, moduleOp);
  mlirRegionAppendOwnedBlock(circuitRegion, circuitBlock);
  mlirOperationStateAddOwnedRegions(&circuitState, 1, &circuitRegion);

  MlirOperation circuit = mlirOperationCreate(&circuitState);
  mlirBlockAppendOwnedOperation(body, circuit);

  mlirOperationDump(mlirModuleGetOperation(module));
  mlirExportFIRRTL(module, &export, NULL);

  mlirModuleDestroy(module);
  mlirContextDestroy(ctx);
  return 0;
}
```

`mlirExportFIRRTL` is a new function, and I will open a PR for it later.

```c
MlirLogicalResult mlirExportFIRRTL(MlirModule module,
                                   MlirStringCallback callback,
                                   void *userData) {
  mlir::detail::CallbackOstream stream(callback, userData);
  return wrap(exportFIRFile(unwrap(module), stream));
}
```

Output:

```
"builtin.module"() ({
  "firrtl.circuit"() ({
    "firrtl.module"() {annotations = []} : () -> ()
  }) {annotations = [], name = "TestCir"} : () -> ()
}) : () -> ()
circuit TestCir :
mlir-c-demo: /circt/llvm/mlir/lib/IR/SymbolTable.cpp:224: static mlir::StringAttr mlir::SymbolTable::getSymbolName(mlir::Operation*): Assertion `name && "expected valid symbol name"' failed.
fish: Job 1, './mlir-c-demo' terminated by signal SIGABRT (Abort)
```

uenoku, 2023-03-02:
firrtl.module doesn't have name attribute. It uses a symbol so you have to add a symbol attribute. Probably it should be like this https://github.com/llvm/llvm-project/blob/569222e172e5d28d66e9607325475b107cee20cb/mlir/test/CAPI/ir.c#L108-L110 ( I'm not familiar with C-API so I might be wrong).

SpriteOvO, 2023-03-02:
@uenoku It works, thank you!

closed 2023-03-02 by SpriteOvO