llvm/circt #4751: Synthesizing generated files with Yosys threw a `unexpected TOK_AUTOMATIC` error

carlosedp, 2023-03-02:
I have a module that was generated with latest Chisel 3.6-RC3 and Firtool 1.32:

```scala
class Blinky(
  freq:    Int,
  startOn: Boolean = false,
) extends Module {
  val io = IO(new Bundle {
    val led0 = Output(Bool())
  })

  // Blink leds every second (start on)
  val led              = RegInit(startOn.B)
  val (_, counterWrap) = Counter(true.B, freq / 2)
  when(counterWrap) {
    led := ~led
  }

  io.led0 := led
}
```

```systemverilog
// Generated by CIRCT firtool-1.32.0
module Blinky(
  input  clock,
         reset,
  output io_led0
);

  reg        led;
  reg [24:0] counterWrap_c_value;
  always @(posedge clock) begin
    if (reset) begin
      led <= 1'h0;
      counterWrap_c_value <= 25'h0;
    end
    else begin
      automatic logic counterWrap = counterWrap_c_value == 25'h17D783F;
      led <= counterWrap ^ led;
      if (counterWrap)
        counterWrap_c_value <= 25'h0;
      else
        counterWrap_c_value <= counterWrap_c_value + 25'h1;
    end
  end // always @(posedge)
  assign io_led0 = led;
endmodule

```

Yosys Error:
```
3. Executing Verilog-2005 frontend: src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/mem_combMem_0.sv
Parsing SystemVerilog input from `src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/mem_combMem_0.sv' to AST representation.
Storing AST representation for module `$abstract\mem_combMem_0'.
Successfully finished Verilog frontend.

4. Executing Verilog-2005 frontend: src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/Blinky.sv

INFO:EdalizeLauncher:Wrapper Command: docker run --rm -v /Users/cdepaula/projects/fusesoc/chiselv/build/carlosedp_chiselv_singlecycle_0:/src -w /src/ulx3s_85-trellis gcr.io/hdl-containers/yosys yosys -l yosys.log -p tcl edalize_yosys_template.tcl
src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/Blinky.sv:16: ERROR: syntax error, unexpected TOK_AUTOMATIC
```

I found https://github.com/llvm/circt/issues/1633 and understood it might have been fixed.

seldridge, 2023-03-02:
Some tools can't support `automatic` (which is unfortunate as it's the only way that I know to have a local temporary without adding a wire to global scope). You can turn this off and get a wire with `--lowering-options=disallowLocalVariables` to avoid this.

carlosedp, 2023-03-02:
Thanks @seldridge! Apparently fixed this issue and now Yosys is complaining about that multiline wire (`expecting TOK_ID`):

```
16. Executing Verilog-2005 frontend: src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/RegisterBank.sv

INFO:EdalizeLauncher:Wrapper Command: docker run --rm -v /Users/cdepaula/projects/fusesoc/chiselv/build/carlosedp_chiselv_singlecycle_0:/src -w /src/ulx3s_85-trellis gcr.io/hdl-containers/yosys yosys -l yosys.log -p tcl edalize_yosys_template.tcl
src/carlosedp_chiselv_singlecycle-ulx3s_0/generated/RegisterBank.sv:47: ERROR: syntax error, unexpected '[', expecting TOK_ID or '#'

ERROR: Failed to build carlosedp:chiselv:singlecycle:0 : '['make']' exited with an error: 2
```

Where the SV is:

```systemverilog
46:   reg  [31:0]       regs_31;
47:   wire [31:0][31:0] _GEN =
48:     {{regs_31},
49:      {regs_30},
50:      {regs_29},
51:      {regs_28},
52:      {regs_27},
53:      {regs_26},
54:      {regs_25},
55:      {regs_24},
56:      {regs_23},
57:      {regs_22},
58:      {regs_21},
59:      {regs_20},
60:      {regs_19},
61:      {regs_18},
62:      {regs_17},
63:      {regs_16},
64:      {regs_15},
65:      {regs_14},
66:      {regs_13},
67:      {regs_12},
68:      {regs_11},
69:      {regs_10},
70:      {regs_9},
71:      {regs_8},
72:      {regs_7},
73:      {regs_6},
74:      {regs_5},
75:      {regs_4},
76:      {regs_3},
77:      {regs_2},
78:      {regs_1},
79:      {regs_0}};
80:   always @(posedge clock) begin
81:     if (reset) begin
82:       regs_0 <= 32'h0;
83:       regs_1 <= 32'h0;

```

If you prefer I can open another issue with the source Chisel and the output SV.

seldridge, 2023-03-02:
Does `--lowering-options=disallowPackedArrays` do it? (I think this is a duplicate of the fixed https://github.com/llvm/circt/issues/1592.)

carlosedp, 2023-03-02:
Yea, it worked fine!
Thanks!!

closed 2023-03-02 by carlosedp