llvm/circt #4752: Memory initialization is generated only inside the `ifndef SYNTHESIS` block

carlosedp, 2023-03-02:
As a similar issue was fixed back in firrtl, memory initialization is currently generated inside the `ifndef SYNTHESIS` block meaning some synthesys tools (like Yosys for example) doesn't pick that up breaking memory initialization on FPGAs.

This was treated in firrtl initially with an annotation and later with a --target:fpga flag which put the initialization block in it's own block after the SYNTHESYS `endif` like:

```
...
`endif // SYNTHESIS
initial begin
  $readmemh("insns.hex", ram);
end
```

What would be the best way to do this? Some flag like the former `--target:fpga`?

References for the issue and PR:

- https://github.com/chipsalliance/firrtl/issues/2114
- https://github.com/chipsalliance/firrtl/pull/2166

jackkoenig, 2023-05-15:
This is similar to @nandor's desire to have register preset work in FPGA synthesis, so perhaps a similar mechanism could be employed for both.

carlosedp, 2023-08-28:
On recent changes, the addition of the `ENABLE_INITIAL_MEM_` guard replaced this `SYNTHESIS` block so adding this define to the synthesis tool (like `verilator -O3 -DENABLE_INITIAL_MEM_ ...` in verilator or -Dname=ENABLE_INITIAL_MEM_ in Yosys) made memory initialization work fine.

carlosedp, 2024-01-08:
Closing as passing defines in synthesis fix this.

closed 2024-01-08 by carlosedp