llvm/circt #4755: The README should explain why are there separate tags for firtool, pycde, sifive

yurivict, 2023-03-02:
Are these separate sub-projects?

If I would like to create ports, should circt be ported as a whole or should firtool, pycde, sifive be ported separately?

mikeurbach, 2023-03-02:
Yes, there are separate subprojects, which tag and release from CIRCT main at different points in time. The `firtool` and `sifive` tags are related to the release cadence and testing for the FIRRTL compiler subproject of CIRCT (the tagging scheme recently changed names from `sifive` to `firtool`). The `pycde` tags are related to the release cadence and testing for the PyCDE subproject of CIRCT.

When you say "create ports", what do you mean by that? For the most part, CIRCT uses trunk based development on main, much like the LLVM project. The different subprojects in the CIRCT codebase tag different commits on main when they are ready to make a release.

dtzSiFive, 2023-03-02:
I believe this is about packaging CIRCT, "ports" are FreeBSD's name for this: https://www.freebsd.org/ports/ .

yurivict, 2023-03-02:
Yes, I would like to create FreeBSD ports for CIRCT.