llvm/circt #4780: [FIRRTL] Missed comb cycle, crash in canonicalizer

dtzSiFive, 2023-03-08:
Input:

```firrtl
circuit wut :
   module wut :
     output rwDataOut : UInt<8>

     rwDataOut <= rwDataOut
```

<details>
<summary>Crashes after making it to HW dialect:</summary>

```
firtool: /home/will/src/sifive/circt/llvm/mlir/include/mlir/IR/UseDefLists.h:190: void mlir::IRObjectWithUseList<mlir::OpOperand>::replaceAllUsesWith(ValueT &&) [OperandType = mlir::OpOperand, ValueT = mlir::Value &]: Assertion `(!newValue || this != OperandType::getUseList(newValue)) && "cannot RAUW a value with itself"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: ./build/bin/firtool wut.fir
 #0 0x00000000006c71bb llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/will/src/sifive/circt/llvm/llvm/lib/Support/Unix/Signals.inc:602:13
 #1 0x00000000006c5450 llvm::sys::RunSignalHandlers() /home/will/src/sifive/circt/llvm/llvm/lib/Support/Signals.cpp:105:18
 #2 0x00000000006c786d SignalHandler(int) /home/will/src/sifive/circt/llvm/llvm/lib/Support/Unix/Signals.inc:0:3
 #3 0x00007f882063dbf0 __restore_rt (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x3dbf0)
 #4 0x00007f882068abc7 __pthread_kill_implementation (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x8abc7)
 #5 0x00007f882063db46 gsignal (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x3db46)
 #6 0x00007f88206284b5 abort (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x284b5)
 #7 0x00007f88206283d9 _nl_load_domain.cold.0 (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x283d9)
 #8 0x00007f88206367b6 (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x367b6)
 #9 0x0000000000a5476c std::enable_if<!(std::is_convertible<mlir::ValueRange&, mlir::Operation*>::value), void>::type mlir::ResultRange::replaceAllUsesWith<mlir::ValueRange&>(mlir::ValueRange&) /home/will/src/sifive/circt/llvm/mlir/include/mlir/IR/ValueRange.h:271:5
#10 0x0000000000a52ec3 mlir::RewriterBase::replaceOp(mlir::Operation*, mlir::ValueRange) /home/will/src/sifive/circt/llvm/mlir/lib/IR/PatternMatch.cpp:275:61
#11 0x0000000000c31cb7 circt::sv::WireOp::canonicalize(circt::sv::WireOp, mlir::PatternRewriter&) /home/will/src/sifive/circt/lib/Dialect/SV/SVOps.cpp:1460:18
#12 0x0000000000e0ccb7 mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<mlir::LogicalResult (mlir::Pattern const&)>) /home/will/src/sifive/circt/llvm/mlir/lib/Rewrite/PatternApplicator.cpp:200:25
#13 0x0000000000dd93ed (anonymous namespace)::GreedyPatternRewriteDriver::processWorklist() /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:229:19
#14 0x0000000000dd702d (anonymous namespace)::RegionPatternRewriteDriver::simplify() && /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:458:15
#15 0x0000000000dd702d mlir::applyPatternsAndFoldGreedily(mlir::Region&, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:487:47
#16 0x0000000000a66148 mlir::LogicalResult::failed() const /home/will/src/sifive/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:44:33
#17 0x0000000000a66148 mlir::applyPatternsAndFoldGreedily(mlir::Operation*, mlir::FrozenRewritePatternSet const&, mlir::GreedyRewriteConfig) /home/will/src/sifive/circt/llvm/mlir/include/mlir/Transforms/GreedyPatternRewriteDriver.h:115:70
#18 0x0000000000a66148 (anonymous namespace)::Canonicalizer::runOnOperation() /home/will/src/sifive/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:61:9
#19 0x0000000000ee9829 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:0:11
#20 0x0000000000ee9e61 mlir::LogicalResult::failed() const /home/will/src/sifive/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:44:33
#21 0x0000000000ee9e61 mlir::failed(mlir::LogicalResult) /home/will/src/sifive/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:72:58
#22 0x0000000000ee9e61 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:534:9
#23 0x0000000000eeb518 std::vector<std::atomic<bool>, std::allocator<std::atomic<bool> > >::size() const /nix/store/z1yw9cw97bcdfmxz5wk7j1bycw42r1mw-gcc-11.3.0/include/c++/11.3.0/bits/stl_vector.h:919:66
#24 0x0000000000eeb518 std::vector<std::atomic<bool>, std::allocator<std::atomic<bool> > >::operator[](unsigned long) /nix/store/z1yw9cw97bcdfmxz5wk7j1bycw42r1mw-gcc-11.3.0/include/c++/11.3.0/bits/stl_vector.h:1045:2
#25 0x0000000000eeb518 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:759:5
#26 0x0000000000eeb518 mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo> > >, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&>(mlir::MLIRContext*, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo> > >, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo> > >, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&) /home/will/src/sifive/circt/llvm/mlir/include/mlir/IR/Threading.h:46:18
#27 0x0000000000eeb518 mlir::LogicalResult mlir::failableParallelForEach<std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo> >&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&>(mlir::MLIRContext*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo> >&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_14&) /home/will/src/sifive/circt/llvm/mlir/include/mlir/IR/Threading.h:92:10
#28 0x0000000000eeb518 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:764:14
#29 0x0000000000ee9865 mlir::detail::OpToOpPassAdaptor::runOnOperation(bool) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:0:5
#30 0x0000000000ee9865 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:468:14
#31 0x0000000000eec3a3 mlir::LogicalResult::failed() const /home/will/src/sifive/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:44:33
#32 0x0000000000eec3a3 mlir::failed(mlir::LogicalResult) /home/will/src/sifive/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:72:58
#33 0x0000000000eec3a3 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:534:9
#34 0x0000000000eec3a3 mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:843:10
#35 0x0000000000eec3a3 mlir::PassManager::run(mlir::Operation*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:823:60
[... cut ...]
#46 0x00007f8820629309 __libc_start_main@GLIBC_2.2.5 (/nix/store/yzjgl0h6a3qh1mby405428f16xww37h0-glibc-2.35-224/lib/libc.so.6+0x29309)
#47 0x0000000000617825 _start (./build/bin/firtool+0x617825)
zsh: IOT instruction (core dumped)  ./build/bin/firtool wut.fir
```

</details>

Dumping the IR shows this before the crash:

```
// -----// IR Dump Before Canonicalizer (canonicalize) //----- //
hw.module @wut() -> (rwDataOut: i8) {
  %.rwDataOut.output = sv.wire : !hw.inout<i8>
  %0 = sv.read_inout %.rwDataOut.output : !hw.inout<i8>
  sv.assign %.rwDataOut.output, %0 : i8
  hw.output %0 : i8
}
```

seldridge, 2023-03-08:
Does the bug also exist for self connected instance inputs?

dtzSiFive, 2023-03-08:
I don't think so, but that's because when I tried it was rejected due to having source flow.  Is there a way to connect an input to itself that I'm forgetting/missing?

seldridge, 2023-03-08:
Something like this. It looks okay, though!

```
# cat Foo.fir | tee /dev/stderr | firtool -format=fir
circuit Foo:
  module Bar:
    input a: UInt<1>
  module Foo:
    inst bar of Bar
    bar.a <= bar.a
<stdin>:4:10: error: detected combinational cycle in a FIRRTL module, sample path: Foo.bar.a <- Foo.bar.a <- Foo.bar.a
  module Foo:
         ^
```

dtzSiFive, 2024-05-17:
This appears to be working now, woohoo!

closed 2024-05-17 by dtzSiFive