llvm/circt #4831: [SCFToCalyx] Use `ref` construct when lowering `memref`

rachitnigam, 2023-03-15:
The current `scf` to `calyx` conversion handles `memref` arguments by directly inlining all the ports for the memory into the component's interface. Instead of doing this, we should use the new [`ref` keyword](https://docs.calyxir.org/lang/memories-by-reference.html) from Calyx and let lower passes handle the transformation of a `ref` cell by inlining its ports into the component.

The `ref` keyword is more general–it allows you to pass in any component by reference into another component to express component sharing–and it allows the compiler to simultaneously generate simulatable and synthesizable code. In the current state of affairs, the compiler cannot simulate designs that only have toplevel `memref`s in them.

Note that this also requires changing the `invoke` construct to accept components by name (see documentation from Calyx above).

rachitnigam, 2023-05-18:
This also requires solving #1679 because the `ref` syntax only works with `invoke`

cgyurgyik, 2025-01-18:
Addressed by #7164.

closed 2025-01-18 by cgyurgyik