llvm/circt #4843: [Lower SCF/Standard to Calyx] Add support for scf::IfOp

xerpi, 2023-03-16:
Hi, I was wondering if it would be a good idea to add lowering support of `scf::IfOp` in `createSCFToCalyxPass` .
`createSCFToCalyxPass` claims to handle `BranchOpInterface` but when it encounters a `scf::YieldOp` it asserts if it's not `scf::WhileOp`.

I've also tried using `createConvertSCFToCFPass` to lower `scf::IfOp` into `cf` and it works well for `scf::IfOp`, but the problem is that it also converts `scf::WhileOp`  into `cf::BranchOp` with backedges, so `createSCFToCalyxPass` fails with:
```
error: CFG backedge detected. Loops must be raised to 'scf.while' or 'scf.for' operations.
```

Here's a `simple_arith_if.mlir`, a port of [simple_arith.mlir](https://github.com/llvm/circt/blob/main/integration_test/Dialect/Calyx/simple_arith.mlir ) to use `scf::IfOp` instead of `cf::CondBranchOp` :
```llvm
func.func @main(%arg0 : i32, %arg1 : i32) -> i32 {
  %0 = arith.cmpi slt, %arg0, %arg1 : i32
  %1 = scf.if %0 -> (i32) {
    scf.yield %arg0 : i32
  } else {
    scf.yield %arg1 : i32
  }
  return %1 : i32
}
```

cgyurgyik, 2023-03-16:
I'm less involved these days, but this makes sense to me. WDYT @mortbopet ?

mortbopet, 2023-03-17:
I think it would be a good idea, and i'd expect that the existing lowering for `BranchOpInterface` probably can be repurposed almost 1:1 (or factored out/generalized/...).

As you mention, there is no good way in MLIR to selectively ask conversion passes to only consider a subset of dialect operations (unless they explicitly implement this) which leads us into situations like this.

rachitnigam, 2023-08-18:
The original program is too simple and can therefore be canonicalized to `arith.select` (which is now supported by the Calyx flow). Here is a slight more complex program for testing an implementation of this feature:
```
func.func @main(%arg0 : i32, %arg1 : i32) -> i32 {
  %0 = arith.cmpi slt, %arg0, %arg1 : i32
  %1 = scf.if %0 -> i32 {
    %3 = arith.addi %arg0, %arg1 : i32
    scf.yield %3 : i32
  } else {
    scf.yield %arg1 : i32
  }
  return %1 : i32
}
```

jiahanxie353, 2023-09-07:
Hi folks, I'd like to give it a shot!
I've looked into the problem of the current pass that it seems to assume `scf.yield` will only occur in `scf.while`, whereas it can actually occur in `scf.if` as well. But I would appreciate some guidance and pointers on how to get started, or if there is any particular consideration I need to be aware of.
Thanks in advance!

jiahanxie353, 2023-10-05:
Hi folks, I've been diving deep into this, but hit a bit of a roadblock. I've created a draft PR to give you a snapshot of where I am, what I've been trying, and the challenges that have me scratching my head.
If you get a chance, could you take a look and shoot over any thoughts or pointers? It'd be super helpful! Thanks a lot!