llvm/circt #4851: Deduplication fails together with dontTouch(io) in example from Chisel

oharboe, 2023-03-18:
@seldridge Will fill out with example....

seldridge, 2023-03-19:
Is there anything going on with either width inference or reset inference occurring differently for different instances? There isn't any optimization or removal of ports before dedup in CIRCT as far as I am aware.

To at least guarantee that deduplication happens (and error if it doesn't), you can use a `firrtl.transforms.MustDeduplicate` annotation to indicate that some instances are expected to all dedup. If this then fails, CIRCT will give you some information as to why.

There's no Chisel API for the annotation, but you can use:

```scala
object mustDedup {
  def apply(modules: RawModule*): Unit = {
    annotate(
      new ChiselAnnotation {
        override def toFirrtl = MustDeduplicateAnnotation(modules.map(_.toTarget))
      }
    )
  }
}
```

oharboe, 2023-03-19:
No error messags or warnings:

```
      val foo = Module(new Foo(i))

      mustDedup(foo)
```

oharboe, 2023-03-19:
I'm passing the reset through the macros right to left which are in an array to avoid a separate top-level reset route to all modules. This means that the leftmost module is different, the reset out isn't used. However, I tried to reproduce this in a small example and deduplication still works.

When I generate .sv files directly from Chisel, there is no .fir file. I guess it is a temporary file? Investigating...

oharboe, 2023-03-19:
Using ```--log-level=info``` I can see that there is no .fir file generated, it seems like the .fir file is just passed directly on stdin.

It would have been useful to have the .fir file for debugging purposes...

```
[info] Running CIRCT: 'firtool -format=fir -warn-on-unprocessed-annotations -dedup -output-annotation-file circt.anno.json --lowering-options=disallowLocalVariables,disallowPackedArrays --split-verilog --dedup --strip-debug-info --extract-test-code [deleted] Foo.anno.json < $input
```

oharboe, 2023-03-19:
I switched back to Chisel verilog generation to get a .fir file. When I compare the modules that are not being de-duplicated, I don't see any differences.

Note that the module to be deduplicated is nesting other modules.

Aha! I found a useful clue. De-duplication doesn't work in Chisel either.

This gives me some hope. I had to modify my code so that it would work with LLVM CIRCT. I should be able to bisect my way to the change that broke deduplication.

oharboe, 2023-03-19:
The problem is that a module where an output is used as a source in BoringUtils.bore() won't be deduplicated.

closed 2023-03-19 by oharboe