llvm/circt #4870: Nightly test failures -- 3/21/23

teqdruid, 2023-03-21:
The nightly tests failed this morning. Looks like timeouts on the gcc release builds.

https://github.com/llvm/circt/actions/runs/4478738645/jobs/7871878079

I suspect something about these builds is causing one of the integration tests to spin forever. Unfortunately, the workflow doesn't specify which. I thought there were timeouts on the integration tests to prevent this from happening and report the tests which caused the timeouts, but I'm either imagining that or it's not working.

dtzSiFive, 2023-03-21:
Bah, thanks for reporting.

Very likely it's this: https://github.com/llvm/circt/commit/6d182f15d1ab8ed9c9cab335ff5193e01b3d2503 .

ASAN reports stack-use-after-scope, sent report to folks, but been in meetings so haven't had a chance to investigate/fix myself.

Put it into a gist for anyone interested: https://gist.github.com/dtzSiFive/91d14ed4e18a1f6c4bfffcccacfaff0b .

dtzSiFive, 2023-03-23:
Yep, they've been passing for the past two nights, closing.  Thanks for reporting!

closed 2023-03-23 by dtzSiFive