llvm/circt #4883: How to avoid Unhandled annotation in Chisel-CIRCT

LandonWong, 2023-03-24:
When I use CIRCT to compile, I got these erri
```
Running CIRCT: 'firtool -format=fir -warn-on-unprocessed-annotations -verify-each=false -dedup -annotation-file build/SimTop.anno.json < $input'
------------------------------------------------------------------------------
Error: firtool failed.
ExitCode:
1
STDOUT:
STDERR:
<stdin>:1:1: error: Unhandled annotation: {class = "freechips.rocketchip.util.ParamsAnnotation", params = {baseAddress = 939524096 : i64, intStages = 0 : i64}, paramsClassName = "freechips.rocketchip.devices.tilelink.CLINTParams", target = "SimTop.CLINT"}
circuit SimTop :
^
<stdin>:1:1: error: Unhandled annotation: {class = "freechips.rocketchip.util.RegFieldDescMappingAnnotation", regMappingSer = {baseAddress = 939524096 : i64, deviceName = "CLINT", 

...

<stdin>:1:1: error: Annotation targets non-wireable operation: {class = "firrtl.passes.wiring.SourceAnnotation", pin = "DISPLAY_LOG_ENABLE", target = "SimTop.SimTop.logEnable"}
circuit SimTop :
^
<stdin>:1:1: error: Unable to apply annotation: {class = "firrtl.passes.wiring.SourceAnnotation", pin = "DISPLAY_LOG_ENABLE", target = "SimTop.SimTop.logEnable"}
circuit SimTop :
^
<stdin>:1:1: error: Unable to resolve source for pin: "DISPLAY_LOG_ENABLE"
circuit SimTop :
^
------------------------------------------------------------------------------
```
It seems like CIRCT cannot recognize these annotation, and I do know CIRCT does not support every annotation and custom transform-backed extension to Chisel. But how to avoid it?

* Chisel Version: 3.5.0
* Chisel-CIRCT: 0.6.0
* llvm/CIRCT: latest

TimeKeepper, 2024-11-03:
Encountered the same issue. Have you found a solution?

seldridge, 2024-11-03:
You can proceed with compilation with `--disable-annotation-unknown` which will just ignore the annotation. The safety of this depends on the annotation. If the annotation is truly optionally discardable, then it's fine. If the annotation is used to guide a custom pass that is mandatory for the correctness of the compilation, then it is not safe. (This is the reason why returning an error is the default behavior.)

`firtool` does have support now for custom passes injected into specific points in the pass pipeline. This is one route. The other route is working to remove the reliance on the annotation by use of properties/object model or seeing if this is a missing first class feature of FIRRTL.