llvm/circt #4889: [PyCDE] Integer outputs broken

teqdruid, 2023-03-25:
```python
from pycde import Input, Output, Module, System
from pycde import generator
from pycde.types import UInt

class AddInts(Module):
  a = Input(UInt(32))
  b = Input(UInt(32))
  c = Output(UInt(33))

  @generator
  def construct(self):
    self.c = self.a + self.b

system = System(AddInts)
system.generate()
system.print()
system.compile()
```

```
module {
  msft.module @AddInts {} (%a: ui32, %b: ui32) -> (c: ui33) attributes {fileName = "AddInts.sv"} {
    %0 = hwarith.add %a, %b {sv.namehint = "a_plus_b"} : (ui32, ui32) -> ui33
    msft.output %0 : ui33
  }
}
Traceback (most recent call last):
  File "/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/docs/PyCDE/examples/basic.py", line 19, in <module>
    system.compile()
  File "/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/.venv/lib/python3.9/site-packages/pycde/system.py", line 314, in compile
    self.run_passes()
  File "/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/.venv/lib/python3.9/site-packages/pycde/system.py", line 291, in run_passes
    pm.run(self.mod.operation)
pycde.circt._mlir_libs.MLIRError: Failure while executing pass pipeline:
error: "/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/docs/PyCDE/examples/basic.py":12:0: null operand found
 note: "/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/docs/PyCDE/examples/basic.py":12:0: see current operation: "msft.output"(<<NULL VALUE>>) : (<<NULL TYPE>>) -> ()
```

teqdruid, 2023-03-25:
@mortbopet: I haven't had time to look into this. I suspect it's not in PyCDE but in one of the lowering passes. Tagging you for visibility.

teqdruid, 2023-03-28:
```mlir
// RUN: circt-opt -lower-hwarith-to-hw %s

msft.module @AddInts {} (%a: ui32, %b: ui32) -> (c: ui33) {
  %0 = hwarith.add %a, %b : (ui32, ui32) -> ui33
  msft.output %0 : ui33
}
```

```
test.mlir:3:5: error: null operand found
    msft.output %0 : ui33
    ^
test.mlir:3:5: note: see current operation: "msft.output"(<<NULL VALUE>>) : (<<NULL TYPE>>) -> ()
```

@mortbopet we [discussed this section of code](https://github.com/llvm/circt/blob/3965f68d8c4633ee3de0db29645699277ce5352d/lib/Conversion/HWArithToHW/HWArithToHW.cpp#L524), which I'm pretty sure is the problem. [Discussion](https://github.com/llvm/circt/pull/4331).

teqdruid, 2023-03-28:
I will fix this in PyCDE by running the hwarith lowering after lowering to hw.module, but we should really fix that lowering. hwarith should be lowerable anywhere it's used.

closed 2023-03-28 by teqdruid

mortbopet, 2023-03-28:
Apologies for the delay.

>  should be lowerable anywhere it's used

Should it? I mean, I principally agree, but how do we ensure this, without having to add every single possible dialect user within the HWArithToHW lowering pass? (and ideally not by requiring all users to implement some HWArith interface).

mortbopet, 2023-03-28:
- Aaand, it's done https://github.com/llvm/circt/pull/4898.

commit 42fd93de (2023-03-28): [PyCDE] Fix lowering for signed math

Fixes #4889.