llvm/circt #4891: [ExportVerilog] Assignment from singed expression to unsigned wires

uenoku, 2023-03-27:
Currently we emit all wires without explicitly denoting signess (e.g. `wire [8:0] foo = `, not `wire singed [8:0] foo`) so the default behavior (I guess) is unsigned. This is causing assignments between signed and unsigned expressions:
```mlir
  hw.module @Top(%a: i8, %b: i8) -> (c: i1) {
    %0 = comb.shrs bin %a, %b : i8
    %1 = comb.extract %0 from 1 : (i8) -> i1
    hw.output %1 : i1
  }
```
Current output:
```verilog
  wire [7:0] _GEN = $signed($signed(a) >>> b);	// bar.mlir:2:10
  assign c = _GEN[1];	// bar.mlir:2:10, :3:10, :4:5
```

Maybe we need to fix into the following form since we need correct widths too?
```verilog
  wire singed [7:0] _GEN = $signed($signed(a) >>> b);	// bar.mlir:2:10
  assign c = _GEN[1];	// bar.mlir:2:10, :3:10, :4:5
```

uenoku, 2023-03-28:
> Just an FYI that sign-extension interplays with width inferencing as well. SFC would always use the $signed expression rather than emit signed wires for this reason.

Thank you for the information! Yeah, that's very interesting. So then the current emission style `wire [7:0] _GEN = $signed(..);` is practically better? There is some tool that complains signed to unsigned assignment so we need to ignore the warning.

azidar, 2023-03-28:
It's definitely easier to emit because you don't need to transform the code to avoid this weird, non-local semantic. And yeah, fundamentally there are tools which have non-overlapping sets of lint rules, meaning almost certainly there does not exist a completely lint free Verilog representation of a design, for all tools.

As general principles for Verilog emission, I tend to order them in the following order of importance:

1.  Verilog subset with local/well understood semantics without corner-cases. This will reduce the likelihood of a semantic mismatch between FIRRTL semantics and Verilog semantics, which a user could randomly encounter which would introduce a bug into their design
2. Lint-free as possible for common tools
3. Verilog readability/debuggability

I'm bringing up this example from deep in my memory, so I'd recommend you look more closely into how signed-ness and width expansion play with each other (read the spec and experiment in a playground). If you can figure out something that works, then go ahead but just know that... here be dragons :)