llvm/circt #4893: [CalyxToHW] Convert low Calyx IR to SSA form instead of emitting `sv` wires

mortbopet, 2023-03-27:
Currently, the CalyxToHW pass takes the least path of resistance in getting from the inline/low Calyx IR form, into something that can be ingested by the SV emitter. This implies that Calyx wires and assigns* are verbatim translated to their SystemVerilog counterparts. The downside of this is that once `sv.wire`s come into play, this heavily reduces the applicability of the canonicalizers of the RTL dialects.

An alternative to this is to convert the low Calyx IR back into SSA form, either before or during CalyxToHW, greatly improving the optimizeability of the lowered design.

*: after guarded statements have been transformed into a mux tree.

mikeurbach, 2023-03-27:
Can you share an example IR that shows the shortcoming of lowering to `sv.wire`? I assume it will be a situation the `sv.wire` canonicalizer can't remove, but I'm curious what Calyx patterns tickle this behavior.

mortbopet, 2023-03-28:
You're right - with 3cf16f3b8b90666a27f0c71ca8428c45953d3976, `sv.wire`s will now only have a single assignment (from a mux tree), which means that they'll get canonicalized away. Given that, and that wires aren't created with symbols, I'm not sure that there is any pattern which actually triggers it!

mortbopet, 2023-07-20:
Going to mark this as closed since @mikeurbach's suggested solutions seems good enough for now.

closed 2023-07-20 by mortbopet