llvm/circt #4896: [FIRRTL] Verify annotation symbol uses!

dtzSiFive, 2023-03-28:
Currently there is no verification that the symbols used by annotations exist at all, nevermind any sense of being appropriate/correct.  This should be fixed.  Example:

```firrtl
module {
  firrtl.circuit "Oops" {
    firrtl.module private @Child() attributes {annotations = [{circt.nonlocal = @nla, class = "circt.test"}]} {
    }
    firrtl.module @Oops() {
      firrtl.instance child @Child()
    }
  }
}
```

Presently, `circt-opt` round-trips this, and `firtool` makes it all the way to `PrefixModules` before something notices.

There may be support for Attribute symbol-use verification, so look for such as that would be ideal.  Otherwise we may need to find a way to expose the uses to the symbol verifier, there are interfaces for exposing subelement's that may be appropriate here.  A final but offhand seemingly less-than-ideal approach to consider would be SymbolUserOpInterface's `verifySymbolUses` (I suppose on all our operations that take annotations).

mikeurbach, 2023-03-28:
> There may be support for Attribute symbol-use verification, so look for such as that would be ideal. Otherwise we may need to find a way to expose the uses to the symbol verifier, there are [...]

This has been a long-time thorn in my side. As far as I know there is no help for Attribute or Type symbol-use verification, so you end up needing to do it in an op's verifier. Hopefully there is a small number of ops that need this right now for attributes. I've wanted to add something like we have for SymbolUserOpInterface, like a SymbolUserType/AttributeInterface that provides the symbol table, but I believe this doesn't exist yet.

mikeurbach, 2023-06-29:
I asked about this here, but didn't get any feedback: https://discourse.llvm.org/t/symbol-user-interface-for-attributes-types/69935