llvm/circt #4897: [HWArith][LowerHWArithToHW] "null operand found" when lowering in `msft.module`

teqdruid, 2023-03-28:
```mlir
// RUN: circt-opt -lower-hwarith-to-hw %s

msft.module @AddInts {} (%a: ui32, %b: ui32) -> (c: ui33) {
  %0 = hwarith.add %a, %b : (ui32, ui32) -> ui33
  msft.output %0 : ui33
}
```

```
test.mlir:3:5: error: null operand found
    msft.output %0 : ui33
    ^
test.mlir:3:5: note: see current operation: "msft.output"(<<NULL VALUE>>) : (<<NULL TYPE>>) -> ()
```

I'm pretty sure this [section of code](https://github.com/llvm/circt/blob/3965f68d8c4633ee3de0db29645699277ce5352d/lib/Conversion/HWArithToHW/HWArithToHW.cpp#L524), which I'm pretty sure is the problem. Ideally, hwarith operations would be lowerable anywhere as well as [un]signed ports on any module.

mortbopet, 2023-03-28:
#4898 should fix this. However, i'm suspecting that there might be an issue in the `msft.module` builder;

running: `circt-opt --lower-hwarith-to-hw --debug` on
```mlir
hw.module @AddInts (%a: ui32, %b: ui32) -> (c: ui33) {
  %0 = hwarith.add %a, %b : (ui32, ui32) -> ui33
  hw.output %0 : ui33
}
```

succeeds. However, running the same on
```mlir
msft.module @AddInts {} (%a: ui32, %b: ui32) -> (c: ui33) {
  %0 = hwarith.add %a, %b : (ui32, ui32) -> ui33
  msft.output %0 : ui33
}
```

fails with an `mlir::Region *mlir::Region::getParentRegion(): Assertion `container && "region is not attached to a container"' failed.` assertion during the debug IR print.

Things do work as expected if we **dont add the --debug flag**:

```mlir
// RUN: circt-opt --lower-hwarith-to-hw %s
msft.module @AddInts {} (%a: ui32, %b: ui32) -> (c: ui33) {
  %0 = hwarith.add %a, %b : (ui32, ui32) -> ui33
  msft.output %0 : ui33
}
// outputs
  msft.module @AddInts {} (%a: i32, %b: i32) -> (c: i33) {
    %false = hw.constant false
    %0 = comb.concat %false, %a : i1, i32
    %false_0 = hw.constant false
    %1 = comb.concat %false_0, %b : i1, i32
    %2 = comb.add %0, %1 : i33
    msft.output %2 : i33
  }
```

I'll go ahead and preliminarily add an issue for this.

mortbopet, 2023-07-20:
Seems to have been fixed.

closed 2023-07-20 by mortbopet