llvm/circt #4915: [FIRRTL] Assertion Failed Compile XiangShan

eastonman, 2023-04-01:
Recently I have been doing work to adapt XiangShan project to CIRCT.
I have encountered the following assertion failed when compiling XiangShan using 

Chisel 3.5.0 
chisel-circt 0.8.0

The executed command is as follows

```
firtool -format=fir -warn-on-unprocessed-annotations -verify-each=false -dedup -output-annotation-file circt.anno.json --disable-annotation-unknown -annotation-file build/SimTop.anno.json < $input
```
The trace is as follows

```

------------------------------------------------------------------------------
Error: firtool failed.
ExitCode:
134
STDOUT:

STDERR:
firtool: ../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:384: void {anonymous}::MutableNLA::setInnerSym(mlir::Attribute, mlir::StringAttr): Assertion `symIdx.count(module) && "Mutable NLA did not contain symbol"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.      Program arguments: firtool -format=fir -warn-on-unprocessed-annotations -verify-each=false -dedup -output-annotation-file circt.anno.json --disable-annotation-unknown -annotation-file build/SimTop.anno.json
 #0 0x0000556f427ebd4f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (.localalias) (/nfs/home/manyang/project/circt/build/bin/firtool+0x12fd4f)
 #1 0x0000556f427e9474 SignalHandler(int) (/nfs/home/manyang/project/circt/build/bin/firtool+0x12d474)
 #2 0x00007f5e18e7b3c0 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x143c0)
 #3 0x00007f5e1893403b raise /build/glibc-sMfBJT/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #4 0x00007f5e18913859 abort /build/glibc-sMfBJT/glibc-2.31/stdlib/abort.c:81:7
 #5 0x00007f5e18913729 get_sysdep_segment_value /build/glibc-sMfBJT/glibc-2.31/intl/loadmsgcat.c:509:8
 #6 0x00007f5e18913729 _nl_load_domain /build/glibc-sMfBJT/glibc-2.31/intl/loadmsgcat.c:970:34
 #7 0x00007f5e18925006 (/lib/x86_64-linux-gnu/libc.so.6+0x34006)
 #8 0x0000556f42c0e75e (anonymous namespace)::MutableNLA::setInnerSym(mlir::Attribute, mlir::StringAttr) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:384:5
 #9 0x0000556f42c0fcb3 (anonymous namespace)::Inliner::renameInstance(llvm::StringRef, circt::firrtl::InstanceOp, circt::firrtl::InstanceOp, circt::firrtl::ModuleNamespace&, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> > const&) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:730:23
#10 0x0000556f42c10616 (anonymous namespace)::Inliner::cloneAndRename(llvm::StringRef, mlir::OpBuilder&, mlir::IRMapping&, mlir::Operation&, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> > const&, llvm::DenseSet<mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void> > const&, circt::firrtl::ModuleNamespace&)::'lambda'(mlir::Operation*)::operator()(mlir::Operation*) const /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:869:21
#11 0x0000556f42c17e81 void llvm::function_ref<void (mlir::Operation*)>::callback_fn<(anonymous namespace)::Inliner::cloneAndRename(llvm::StringRef, mlir::OpBuilder&, mlir::IRMapping&, mlir::Operation&, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> > const&, llvm::DenseSet<mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void> > const&, circt::firrtl::ModuleNamespace&)::'lambda'(mlir::Operation*)>(long, mlir::Operation*) /nfs/home/manyang/project/circt/build/../llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:46:40
#12 0x0000556f428a894b llvm::function_ref<void (mlir::Operation*)>::operator()(mlir::Operation*) const /nfs/home/manyang/project/circt/build/../llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:62
#13 0x0000556f428a24fc void mlir::detail::walk<mlir::ForwardIterator>(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, mlir::WalkOrder) /nfs/home/manyang/project/circt/build/../llvm/mlir/include/mlir/IR/Visitors.h:177:45
#14 0x0000556f42c161b8 _ZN4mlir6detail4walkILNS_9WalkOrderE0ENS_15ForwardIteratorEZN12_GLOBAL__N_17Inliner14cloneAndRenameEN4llvm9StringRefERNS_9OpBuilderERNS_9IRMappingERNS_9OperationERKNS6_8DenseMapINS_9AttributeESF_NS6_12DenseMapInfoISF_vEENS6_6detail12DenseMapPairISF_SF_EEEERKNS6_8DenseSetISF_SH_EERN5circt6firrtl15ModuleNamespaceEEUlPSC_E_SW_vEENSt9enable_ifIXsrSt11disjunctionIJSt7is_sameIT2_SW_ES10_IS11_PNS_6RegionEES10_IS11_PNS_5BlockEEEE5valueET3_E4typeESW_OT1_ /nfs/home/manyang/project/circt/build/../llvm/mlir/include/mlir/IR/Visitors.h:313:78
#15 0x0000556f42c157ac _ZN4mlir9Operation4walkILNS_9WalkOrderE0ENS_15ForwardIteratorEZN12_GLOBAL__N_17Inliner14cloneAndRenameEN4llvm9StringRefERNS_9OpBuilderERNS_9IRMappingERS0_RKNS6_8DenseMapINS_9AttributeESE_NS6_12DenseMapInfoISE_vEENS6_6detail12DenseMapPairISE_SE_EEEERKNS6_8DenseSetISE_SG_EERN5circt6firrtl15ModuleNamespaceEEUlPS0_E_vEENSt9enable_ifIXeqsrNS6_15function_traitsINSt5decayIT1_E4typeEXsrSt8is_classIS12_E5valueEEE8num_argsLi1EET2_E4typeEOS10_ /nfs/home/manyang/project/circt/build/../llvm/mlir/include/mlir/IR/Operation.h:641:75
#16 0x0000556f42c10986 (anonymous namespace)::Inliner::cloneAndRename(llvm::StringRef, mlir::OpBuilder&, mlir::IRMapping&, mlir::Operation&, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> > const&, llvm::DenseSet<mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void> > const&, circt::firrtl::ModuleNamespace&) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:877:28
#17 0x0000556f42c11b47 (anonymous namespace)::Inliner::inlineInto(llvm::StringRef, mlir::OpBuilder&, mlir::IRMapping&, circt::BackedgeBuilder&, llvm::SmallVectorImpl<circt::Backedge>&, circt::firrtl::FModuleOp, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> >&, circt::firrtl::ModuleNamespace&) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:1031:21
#18 0x0000556f42c1245b (anonymous namespace)::Inliner::inlineInto(llvm::StringRef, mlir::OpBuilder&, mlir::IRMapping&, circt::BackedgeBuilder&, llvm::SmallVectorImpl<circt::Backedge>&, circt::firrtl::FModuleOp, llvm::DenseMap<mlir::Attribute, mlir::Attribute, llvm::DenseMapInfo<mlir::Attribute, void>, llvm::detail::DenseMapPair<mlir::Attribute, mlir::Attribute> >&, circt::firrtl::ModuleNamespace&) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:1101:17
#19 0x0000556f42c13000 (anonymous namespace)::Inliner::inlineInstances(circt::firrtl::FModuleOp) /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:1198:17
#20 0x0000556f42c1448b (anonymous namespace)::Inliner::run() /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:1308:3
#21 0x0000556f42c14ef4 (anonymous namespace)::InlinerPass::runOnOperation() /nfs/home/manyang/project/circt/build/../lib/Dialect/FIRRTL/Transforms/ModuleInliner.cpp:1440:5
#22 0x0000556f435cdd16 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (.localalias) (/nfs/home/manyang/project/circt/build/bin/firtool+0xf11d16)
#23 0x0000556f435ce2ca mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (.localalias) (/nfs/home/manyang/project/circt/build/bin/firtool+0xf122ca)
#24 0x0000556f435cee9c mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&)::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const (/nfs/home/manyang/project/circt/build/bin/firtool+0xf12e9c)
#25 0x0000556f435cd16f mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (.localalias) (/nfs/home/manyang/project/circt/build/bin/firtool+0xf1116f)
[... cut ...]
#32 0x0000556f4271b6f3 executeFirtool(mlir::MLIRContext&) /nfs/home/manyang/project/circt/build/../tools/firtool/firtool.cpp:936:13
#33 0x0000556f4271b94c main /nfs/home/manyang/project/circt/build/../tools/firtool/firtool.cpp:995:39
#34 0x00007f5e189150b3 __libc_start_main /build/glibc-sMfBJT/glibc-2.31/csu/../csu/libc-start.c:342:3
#35 0x0000556f427184ae _start (/nfs/home/manyang/project/circt/build/bin/firtool+0x5c4ae)
```

The input file SimTop.anno.json is as follows
[SimTop.anno.txt](https://github.com/llvm/circt/files/11128733/SimTop.anno.txt)

Related Issue from XiangShan: OpenXiangShan/XiangShan#1992

prithayan, 2023-04-03:
Can you please share the input `fir` file as well ?

eastonman, 2023-04-03:
https://transfer.sh/WBgSiQ/SimTop.fir.txt

eastonman, 2023-04-03:
I am not familiar with CIRCT or LLVM or FIRRTL.
My colleague is working on some fixes on XiangShan's side, OpenXiangShan/XiangShan#2023.
Not sure whether the problem is from our side or not.

dtzSiFive, 2023-04-03:
FWIW, used `circt-reduce` on these inputs to produce the following reproducer which looks similar to #4920 regarding inlining a module where a `hw.hierpath` starts but doesn't seem to relate to symbol renaming (?).

```mlir
module {
  firrtl.circuit "SimTop" {
    hw.hierpath private @nla_61 [@Rob::@difftest_3, @DifftestLoadEvent]
    hw.hierpath private @nla_60 [@Rob::@difftest_2, @DifftestLoadEvent]
    firrtl.extmodule private @DifftestIntWriteback(in io_clock: !firrtl.clock, in io_coreid: !firrtl.uint<8>, in io_valid: !firrtl.uint<1>, in io_dest: !firrtl.uint<32>, in io_data: !firrtl.uint<64>) attributes {defname = "DifftestIntWriteback"}
    firrtl.extmodule private @DifftestLoadEvent(in io_clock: !firrtl.clock, in io_coreid: !firrtl.uint<8>, in io_index: !firrtl.uint<8>, in io_valid: !firrtl.uint<1>, in io_paddr: !firrtl.uint<64>, in io_opType: !firrtl.uint<8>, in io_fuType: !firrtl.uint<8>) attributes {annotations = [{circt.nonlocal = @nla_60, class = "firrtl.transforms.BlackBoxInlineAnno", name = "DifftestLoadEvent.v", text = "\0Amodule DifftestLoadEvent(\0A  input         io_clock,\0Ainput  [ 7:0] io_coreid,\0Ainput  [ 7:0] io_index,\0Ainput         io_valid,\0Ainput  [63:0] io_paddr,\0Ainput  [ 7:0] io_opType,\0Ainput  [ 7:0] io_fuType\0A);\0A`ifndef SYNTHESIS\0A`ifdef DIFFTEST\0A\0Aimport \22DPI-C\22 function void v_difftest_LoadEvent (\0Ainput     byte io_coreid,\0Ainput     byte io_index,\0Ainput      bit io_valid,\0Ainput  longint io_paddr,\0Ainput     byte io_opType,\0Ainput     byte io_fuType\0A);\0A\0A  always @(posedge io_clock) begin\0A    v_difftest_LoadEvent (io_coreid,io_index,io_valid,io_paddr,io_opType,io_fuType);\0A  end\0A`endif\0A`endif\0Aendmodule\0A"}, {circt.nonlocal = @nla_61, class = "firrtl.transforms.BlackBoxInlineAnno", name = "DifftestLoadEvent.v", text = "\0Amodule DifftestLoadEvent(\0A  input         io_clock,\0Ainput  [ 7:0] io_coreid,\0Ainput  [ 7:0] io_index,\0Ainput         io_valid,\0Ainput  [63:0] io_paddr,\0Ainput  [ 7:0] io_opType,\0Ainput  [ 7:0] io_fuType\0A);\0A`ifndef SYNTHESIS\0A`ifdef DIFFTEST\0A\0Aimport \22DPI-C\22 function void v_difftest_LoadEvent (\0Ainput     byte io_coreid,\0Ainput     byte io_index,\0Ainput      bit io_valid,\0Ainput  longint io_paddr,\0Ainput     byte io_opType,\0Ainput     byte io_fuType\0A);\0A\0A  always @(posedge io_clock) begin\0A    v_difftest_LoadEvent (io_coreid,io_index,io_valid,io_paddr,io_opType,io_fuType);\0A  end\0A`endif\0A`endif\0Aendmodule\0A"}], defname = "DifftestLoadEvent"}
    firrtl.module private @Rob() attributes {annotations = [{class = "firrtl.passes.InlineAnnotation"}]} {
      %difftest_2_io_clock, %difftest_2_io_coreid, %difftest_2_io_index, %difftest_2_io_valid, %difftest_2_io_paddr, %difftest_2_io_opType, %difftest_2_io_fuType = firrtl.instance difftest_2 sym @difftest_2 @DifftestLoadEvent(in io_clock: !firrtl.clock, in io_coreid: !firrtl.uint<8>, in io_index: !firrtl.uint<8>, in io_valid: !firrtl.uint<1>, in io_paddr: !firrtl.uint<64>, in io_opType: !firrtl.uint<8>, in io_fuType: !firrtl.uint<8>)
      %invalid_ui4 = firrtl.invalidvalue : !firrtl.uint<4>
      firrtl.connect %difftest_2_io_fuType, %invalid_ui4 : !firrtl.uint<8>, !firrtl.uint<4>
      %difftest_3_io_clock, %difftest_3_io_coreid, %difftest_3_io_index, %difftest_3_io_valid, %difftest_3_io_paddr, %difftest_3_io_opType, %difftest_3_io_fuType = firrtl.instance difftest_3 sym @difftest_3 @DifftestLoadEvent(in io_clock: !firrtl.clock, in io_coreid: !firrtl.uint<8>, in io_index: !firrtl.uint<8>, in io_valid: !firrtl.uint<1>, in io_paddr: !firrtl.uint<64>, in io_opType: !firrtl.uint<8>, in io_fuType: !firrtl.uint<8>)
      %invalid_ui8 = firrtl.invalidvalue : !firrtl.uint<8>
      firrtl.strictconnect %difftest_3_io_coreid, %invalid_ui8 : !firrtl.uint<8>
    }
    firrtl.module private @CtrlBlock() attributes {annotations = [{class = "firrtl.passes.InlineAnnotation"}]} {
      firrtl.instance rob @Rob()
    }
    firrtl.module @SimTop() {
      %difftest_3_io_clock, %difftest_3_io_coreid, %difftest_3_io_valid, %difftest_3_io_dest, %difftest_3_io_data = firrtl.instance difftest_3 sym @difftest_3 @DifftestIntWriteback(in io_clock: !firrtl.clock, in io_coreid: !firrtl.uint<8>, in io_valid: !firrtl.uint<1>, in io_dest: !firrtl.uint<32>, in io_data: !firrtl.uint<64>)
      %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
      firrtl.strictconnect %difftest_3_io_valid, %invalid_ui1 : !firrtl.uint<1>
      firrtl.instance ctrlBlock @CtrlBlock()
    }
  }
}
```

This produces the error mentioned in the issue with 1.37.0.  In 1.35.0, there is no assertion failure but the verifier reports:
```
<unknown>:0: error: 'hw.hierpath' op  module: "CtrlBlock" does not contain any instance with symbol: "difftest_3"
<unknown>:0: note: see current operation: "hw.hierpath"() {namepath = [#hw.innerNameRef<@CtrlBlock::@difftest_3>, @DifftestLoadEvent], sym_name = "nla_61", sym_visibility = "private"} : () -> ()
<unknown>:0: error: 'hw.hierpath' op  module: "CtrlBlock" does not contain any instance with symbol: "difftest_2"
<unknown>:0: note: see current operation: "hw.hierpath"() {namepath = [#hw.innerNameRef<@CtrlBlock::@difftest_2>, @DifftestLoadEvent], sym_name = "nla_60", sym_visibility = "private"} : () -> ()
```

Tang-Haojin, 2023-04-04:
> I am not familiar with CIRCT or LLVM or FIRRTL. My colleague is working on some fixes on XiangShan's side, [OpenXiangShan/XiangShan#2023](https://github.com/OpenXiangShan/XiangShan/pull/2023). Not sure whether the problem is from our side or not.

It works for me on the recently merged XiangShan master, with [sifive-v0.1.0](https://github.com/llvm/circt/releases/tag/sifive%2F0%2F1%2F0). Maybe you can have a try :)
Currently, perf counters and some other functions using boring util in chisel are unsupported in XiangShan. There are still many works to do to support circt fully.

dtzSiFive, 2023-04-11:
Should be fixed by #4942, please re-open or file a new issue if not resolved on latest git!  Not sure if will make 1.38.0 or if will be in 1.39.0.

closed 2023-04-11 by dtzSiFive