llvm/circt #4933: Out-of-Tree Passes in `firtool`

abejgonzalez, 2023-04-04:
Hi All,

As a prior Scala FIRRTL compiler (SFC) user, I (and many others) have a lot of out-of-tree SFC passes that we use to manipulate module hierarchy and more. We would like to port some of these passes to the new CIRCT/`firtool` compiler but we would like them to live out-of-tree (like the prior SFC approach) instead of having to clone all of CIRCT/LLVM and building the project from scratch. **How much effort would it be to support out-of-tree passes in CIRCT? Is supporting out-of-tree passes on the roadmap?** I'm aware that LLVM seems to have this functionality itself (https://github.com/banach-space/llvm-tutor/tree/main). Now that SFC is being deprecated/removed, it would be nice to have this option, and would help ease the transition from SFC to `firtool`.

**Extra Information/Examples/Context:**
Existing Chipyard project SFC invocation before `firtool` to run out-of-tree SFC passes: https://github.com/ucb-bar/chipyard/blob/afd91d12b536db0cbf6a020d0a4293df329c99f7/common.mk#L195-L231
Existing Chipyard out-of-tree passes: https://github.com/ucb-bar/barstools/tree/beagle-chip/tapeout/src/main/scala/transforms

**Potential List of People Interested In This Work:**
I've had this discussion with a few folks but I'm sure more folks might be interested in this. Here is a non-exhaustive list of people that might have an interest in this.

@joey0320
@sagark
@jerryz123
@harrisonliew
@ekiwi
@vighneshiyer
@tymcauley

mikeurbach, 2023-04-04:
Thanks for opening this. It is on the roadmap, but I'm not sure if we have any other ticket about it yet. What this will look like its the guts of `firtool` that implement the pass pipeline will be exposed, so you can create your own `firtool`-like binary which uses the same core pass pipeline, but with customizations at various points.

mikeurbach, 2023-04-04:
I believe @youngar already started looking at this, so assigning the ticket to him.

uenoku, 2023-04-05:
> I'm aware that LLVM seems to have this functionality itself 

Unfortunately MLIR doesn't have a mechanism to load pass plugins dynamically right now. Though there seems to be an on going work (https://discourse.llvm.org/t/rfc-load-pass-plugin-for-mlir-opt/65049/) but that functionality will not be merged in the near future. So practical solution would be to expose firtool pipeline as a C++ library and create your own firtool-like binary (as Mike suggested). 

Here is old discussion: https://discourse.llvm.org/t/writing-custom-firrtl-passes/62216/2

darthscsi, 2023-04-05:
The annotation lowering was written with the expectation that it would support dynamically loaded passes handling annotations.  I'm assuming some of the customization is annotation driven (which is a bad idea, but one which we have to support for a while).

The two obvious paths under current implementation are:
1) custom tool.  As Mike mentioned.  There is existing desire to pull the firrtl pipeline out to a library so multiple tools can use it.  This would make building a custom firrtl implementation easier.
2) split compilation.  Firtool -ir-fir -> circt-opt (or custom tool) -> firtool -verilog.

The second case is probably the easiest.

dtzSiFive, 2023-04-07:
I like (2) and have used in a different project.  We could optionally drive this ourselves at extension points (pipe through a specified tool/pipeline), although making `firtool` easy to dump and continue from a pipeline point would mean a user could do that themselves.
(2) is easy to implement and doesn't require the transforms be written in C++ / into-a-suitable-shared-library.  The tool could be a simple bash or python script, which can really help give an idea a try.  Of course even if using bytecode, there'll be a bit of serialization/deserialization overhead.

Anyway just noticed that MLIR just landed support for pass and dialect plugins for `mlir-opt`: https://reviews.llvm.org/D147053 so that could be part of a solution.

seldridge, 2023-10-19:
This should be supported now thanks to work by @uenoku!

See:
  - Custom passes via pass plugins: https://github.com/llvm/circt/pull/6254
  - Custom annotation handlers: https://github.com/llvm/circt/pull/6296

If embarking down this path, you will likely need to have CI/CD setup in the downstream (`ucb-bar/chipyard`) to track the upstream project (`llvm/circt`). I have some GitHub Actions infrastructure to help with this:

  - https://github.com/circt/install-circt installs release or nightly versions of CIRCT. This is typically used in both normal CI and nightly CI.
  - https://github.com/circt/update-circt does dependabot-style updating of a downstream project to use a newly released version of CIRCT.

`chipsalliance/chisel` is setup to use these and may serve as a useful reference.

uenoku, 2023-10-19:
Yes once infrastructure got stable I'll update a project template(https://github.com/uenoku/firtool-standalone-plugin) and move the repo somewhere more appropriate(under circt or chipsalliance?)

PR #6296 (2023-10-23): [FIRRTL] LowerAnnotations: Allow annotationRecords to be extended externally 
This PR factors out some of LowerAnnotation helpers to a header and introduces registerAnnotationRecord method so that downstream user can extend the table. 

Fixes #4933.
commit c1d84a85 (2023-10-23): [FIRRTL] LowerAnnotations: Allow annotationRecords to be extended externally  (#6296)

This PR factors out some of LowerAnnotation helpers to a header and introduces registerAnnotationRecord method so that downstream user can extend the table. 

Fixes #4933.
closed 2023-10-23 by uenoku